modelica/ModelicaStandardLibrary #2766: Issues regarding DC machines

max-privato, 2018-11-05:
Consider in Modelica.Electrical.Machines.BasicMachines
DC_ElectricalExcited and DC_SeriesExcited.

The documentation says that DC_SeriesExcited must be used for series connected machines, but the model allows different types of excitation connection. 
In principle models allow exciting DC_SeriesExcited  shunt and DC_ElectricalExcited in series. 

So the following question arises: if DC_SeriesExcited can be used only with the excitation coil in series, why has it four pins, thus allowing also other connections (e.g. in parallel) which give unpredictable results?

I also checked that if DC_ElectricalExcited machine excitation coil is connected in series the results do not correspond to an actual series excited machine.

So I propose
1) to modify the DC_seriesExcited so that the series connection between excitation armature coils is internally made and shown in the machine symbol, so that the machine will have two terminals. Unless there is a good reason for leaving four coils, and in this case this should be explained in the documentation
2) explicitly state in the documentation of DC_ElectricalExctited that the excitation coil cannot be connected in series, giving also a short explanation why.

AHaumer, 2018-11-05:
The ElectricalExcited can be used for separate excitation or parallel excitation. 
The SerialExcited shall be used onyl for serial excitation.
The difference between both models is the calculation of parameter ViNominal (induced voltage) and psi_eNominal (magnetic flux) from nominal parameters (like resistances and nominal currents):
ElectricalExcited:
  ViNominal=VaNominal - Ra*IaNominal
  psi_eNominal=Lme*abs(IeNominal)
SeriesExcited:
  ViNominal=VaNominal - (Ra + Re)*IaNominal
  psi_eNominal=Lme*abs(IaNominal)

Your proposal 1. doesn't make sense: How do you change direction of rotation of a series ecxited machine? You have to change the direction of current through either the armature or the series excitation winding. How should I do that with permanet internally connected pins?
Proposal 2. in favour. Explanation: see above (voltage drop between voltage source and induced voltage for a serial excitation takes additionally the excitation resistance into account). Sorry if you create the model you bear all that in mind, forgetting to explain that. Unfortunately I don't have the time to enhance the documentation within short.

max-privato, 2018-11-05:
@AHaumer, 
thank you for your fast answer.
Your explanation is so clear now. Indeed it answers both proposal 1 (the last phrase of it) and proposal 2.
I myself could have envisaged the answer but I did not dig into code, since I wanted to put myself in the final user position, such as my student's.
Thanks again. 
I suppose it is a good idea to keep this ticket open until the documentation enhancement is done.