modelica/ModelicaStandardLibrary #2776: Modelica.Mechanics.MultiBody.Frames.from_nxz calculation of w

GarronFish, 2018-11-19:
Modelica.Mechanics.MultiBody.Frames.from_nxz sets R.w={0,0,0}
this seems incorrect.  Shouldn't R.w be calculated?

R.w can be calculated with der(T) and T.

GarronFish, 2018-11-19:
Thanks, for pointing that out.  

A non-fixed version would be helpful.

beutlich, 2018-11-19:
> A non-fixed version would be helpful.

Yes, can even be done in a backward-compatible way.

beutlich, 2019-06-30:
Reading section 3.10.6 of the Dymola 2020 Release Notes, I see that you meanwhile worked on this issue.

> Added function from_nxy_movingAxes for calculating the orientation R based on moving axes.

@GarronFish @mdempse1 @Claytex Do you think you could contribute to the MSL this time and provide this enhanced function implementation (either as PR or as code snippet)? Thank you.

GarronFish, 2019-07-01:
Happy to.  I'm not sure how this function would be backwards compatible as the derivatives of the n_x and n_y are required in the calculation.  I was thinking that the user would supply the derivatives n_x_der, and n_y_der (what is the MSL convention for the naming the derivatives of signals?).

When would you like the update by?

beutlich, 2019-07-01:
I thought it would be possible like

```mo
function from_nxz "Return fixed orientation object from n_x and n_z vectors"
  extends Modelica.Icons.Function;
  input Real n_x[3](each final unit="1") "Vector in direction of x-axis of frame 2, resolved in frame 1";
  input Real n_z[3](each final unit="1") "Vector in direction of z-axis of frame 2, resolved in frame 1";
  input Boolean movingAxis = false "= true, if moving axis are to be considered, otherwise false";
  output Orientation R "Orientation object to rotate frame 1 into frame 2";
algorithm
  if movingAxis then
    R := TODO;
  else
    R := Orientation(T=TransformationMatrices.from_nxz(n_x,n_z),w= zeros(3));
  end if;
end from_nxz;
```
Don't you think so?

> When would you like the update by?

Anytime soon. 😄

GarronFish, 2019-07-01:
Unfortunately the derivatives of the axes have to be calculated to determine R.w, this requires derivative information.

I'm unfamiliar with Git which would be a slowing factor.  I could send the function by Wednesday, not sure how long it would take for me to put it into the MSL Git repository, probably by next Monday.  I'd be interested to learn this though.

beutlich, 2019-07-01:
Hm, can't the derivatives be calculated within the function or provided as optional arguments, too?

Regarding git we collected some information in our wiki:
* https://github.com/modelica/ModelicaStandardLibrary/wiki/GUI-Clients-for-Windows
* https://github.com/modelica/ModelicaStandardLibrary/wiki/Git-issues-from-svn-perspective
* https://github.com/modelica/ModelicaStandardLibrary/wiki/References,-how-tos-and-tutorials-to-git

GarronFish, 2019-07-01:
It is not legal to use der() inside a function.

Thanks I'll have a look at those links.

GarronFish, 2019-07-08:
Sorry I'm going to be quite a while, work has just got hectic.  I can provide you with the models soon, if you want to put them into push these onto the Modelica Library.  Otherwise it could be a few weeks before it gets loaded up.

tobolar, 2021-01-18:
@GarronFish Any update on this issue?

GarronFish, 2021-01-19:
Sorry I've just been too busy, here are the functions please incorporate these in Modelica Standard Library
Claytex.Mechanics.MultiBody.Frames.TransformationMatrices.from_nxy_movingAxes:
`function from_nxy_movingAxes
  "Return orientation object from n_x and n_y vectors that are moving"

  input Real n_x[3](each final unit="1")
    "Vector in direction of x-axis of frame 2, resolved in frame 1";
  input Real n_y[3](each final unit="1")
    "Vector in direction of y-axis of frame 2, resolved in frame 1";
  input Real der_n_x[3](each final unit="1/s") = {0,0,0}
    "Velocity of of x-axis of frame 2, resolved in frame 1";
  input Real der_n_y[3](each final unit="1/s") = {0,0,0}
    "Velocity of y-axis of frame 2, resolved in frame 1";

  output MB.Frames.TransformationMatrices.Orientation T
    "Orientation object to rotate frame 1 into frame 2";
  output SI.AngularVelocity w[3]
    "Absolute angular velocity of local frame, resolved in local frame";

protected 
  Real abs_n_x=sqrt(n_x*n_x);
  Real e_x[3](each final unit="1") = n_x/abs_n_x;
  Real abs_n_y=sqrt(n_y*n_y);
  Real e_y[3](each final unit="1") = n_y/abs_n_y;
  Real e_z[3](each final unit="1") = cross(e_x, e_y);

algorithm 
  T := {e_x,e_y,e_z};
  w[1] := e_z*der_n_y/abs_n_y;
  w[2] := -e_z*der_n_x/abs_n_x;
  w[3] := e_y*der_n_x/abs_n_x;

  annotation (smoothOrder=3, Documentation(info="<html>
<p>Similar to <a href=\"modelica://Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.from_nxy\">from_nxy</a> from the Modelica Standard Library except designed for moving axes, this means that the rotational velocity of the axes (i.e. <b>w</b>) is calculated.</p>
<p>It is assumed that the two input vectors n_x and n_y are resolved in frame 1 and are directed along the x and y axis of frame 2 and that they are orthogonal to each other. </p>
<p><b>NOTE:</b> The axes are assumed to be orthogonal this is different from the Modelica Standard Library version which can tolerate n_x and n_y not being orthogonal.</p>
</html>",
revisions="<html>
<table cellspacing=\"0\" cellpadding=\"0\" border=\"0\"><tr>
<td><p>[image]&nbsp;&nbsp;&nbsp;</p></td>
<td><p><br><b>Copyright &copy; 2008-2020, Claytex Services Limited</b></p></td>
</tr>
</table>
</html>"));
end from_nxy_movingAxes;`

Claytex.Mechanics.MultiBody.Frames.from_nxy_movingAxes:
`function from_nxy_movingAxes
  "Return orientation object from moving n_x and n_y axes"

  input Real n_x[3](each final unit="1")
    "Vector in direction of x-axis of frame 2, resolved in frame 1";
  input Real n_y[3](each final unit="1")
    "Vector in direction of y-axis of frame 2, resolved in frame 1";
  input Real der_n_x[3](each final unit="1/s") = {0,0,0}
    "Velocity of x-axis of frame 2, resolved in frame 1";
  input Real der_n_y[3](each final unit="1/s") = {0,0,0}
    "Velocity of y-axis of frame 2, resolved in frame 1";

  output MB.Frames.Orientation R
    "Orientation object to rotate frame 1 into frame 2";
protected 
  MB.Frames.TransformationMatrices.Orientation T
    "Orientation object to rotate frame 1 into frame 2";
  SI.AngularVelocity w[3]
    "Absolute angular velocity of local frame, resolved in local frame";
algorithm 
  (T,w) :=
    Claytex.Mechanics.MultiBody.Frames.TransformationMatrices.from_nxy_movingAxes(
            n_x=n_x,
            n_y=n_y,
            der_n_x=der_n_x,
            der_n_y=der_n_y);

  R := Modelica.Mechanics.MultiBody.Frames.from_T(T=T, w=w);

  annotation (
    smoothOrder=3,
    Documentation(info="<html>
<p>Similar to <a href=\"modelica://Modelica.Mechanics.MultiBody.Frames.TransformationMatrices.from_nxy\">from_nxy</a> from the Modelica Standard Library except designed for moving axes, this means that the rotational velocity of the axes (i.e. <b>R.w</b>) is calculated.</p>
<p>It is assumed that the two input vectors n_x and n_y are resolved in frame 1 and are directed along the x and y axis of frame 2 and that they are orthogonal to each other. </p>
<p><b>NOTE:</b> The axes are assumed to be orthogonal this is different from the Modelica Standard Library version which can tolerate n_x and n_y not being orthogonal.</p>
</html>",
  revisions="<html>
<table cellspacing=\"0\" cellpadding=\"0\" border=\"0\"><tr>
<td><p>[image]&nbsp;&nbsp;&nbsp;</p></td>
<td><p><br><b>Copyright &copy; 2008-2020, Claytex Services Limited</b></p></td>
</tr>
</table>
</html>"),
    Icon(graphics));
end from_nxy_movingAxes;
`
Imports used in the Claytex package:
  `import Modelica.Units.SI;
  import MB = Modelica.Mechanics.MultiBody;`

GarronFish, 2021-01-19:
The documentation annotation unfortunately is being modified by the GitHub viewer.

tobolar, 2021-01-20:
> Sorry I've just been too busy, 

No matter, I bet we all are continuously busy - in spite of Corona.. ;-)