modelica/ModelicaStandardLibrary #2784: Analysis of ModelicaTest v3.2.3-rc.1+71590c7 regressions vs. v3.2.2+build.0-beta.3

beutlich, 2018-11-21:
This is the public [report](https://htmlpreview.github.io/?https://github.com/beutlich/MSLTEST323FAILED/blob/master/index.html) of the 16 failed test models for (untagged) MSL v3.2.3-rc.1+[71590c7](https://github.com/modelica/ModelicaStandardLibrary/commit/71590c7e658a6fbbe037364f52b82589b01a1003)  vs. v3.2.2+build.0-beta.3 with my analysis.

### ModelicaTest\Media\TestAllProperties\IncompleteMedia\ReferenceMoistAir\ReferenceMoistAir.csv
### ModelicaTest\Media\TestAllProperties\MoistAir\MoistAir.csv

The offset in ModelicaTest.Media.TestAllProperties.IncompleteMedia.ReferenceMoistAir and ModelicaTest.Media.TestAllProperties.IncompleteMedia.ReferenceMoistAir is due to the correction of the molar mass in #2378 and acceptable.

@hubertus65 Please confirm.

### ModelicaTest\MultiBody\Forces\Spring2\Spring2.csv
### ModelicaTest\MultiBody\Forces\Torque\Torque.csv
### ModelicaTest\MultiBody\Forces\Torque2\Torque2.csv
### ModelicaTest\MultiBody\Forces\Torque3\Torque3.csv
### ModelicaTest\MultiBody\Joints\JointUSP2\JointUSP2.csv
### ModelicaTest\MultiBody\Parts\Rotor1D\Rotor1D_1\Rotor1D_1.csv

The deviation of the above 6 MultiBody test models is acceptable. @tobolar Please confirm.

### ModelicaTest\MultiBody\PlanarLoopWithMove\PlanarLoopWithMove.csv

@GallLeo See https://github.com/modelica/ModelicaStandardLibrary/issues/2728#issuecomment-430004419.

### ModelicaTest\MultiBody\Joints\RevolutePlanarLoopConstraint\RevolutePlanarLoopConstraint.csv
### ModelicaTest\MultiBody\Joints\UniversalSpherical\UniversalSpherical.csv
### ModelicaTest\MultiBody\Parts\FreeBodies\FreeBodies.csv
### ModelicaTest\MultiBody\Parts\FreeBodies2\FreeBodies2.csv
### ModelicaTest\MultiBody\Parts\FreeBodiesEnforceStatesFalse\FreeBodiesEnforceStatesFalse.csv
### ModelicaTest\MultiBody\Parts\Rotor1D\GearConstraint3\GearConstraint3.csv

@tobolar Please add your analysis of the MultiBody test models here. Clone https://github.com/beutlich/MSLTEST323FAILED locally to get the reports.

### ModelicaTest\Tables\CombiTimeTable\Test6\Test6.csv

This looks like a false positive. Whereas for the MSL v3.2.2 regression run the stop time event was stored as
```csv
...
2.499,2.6,-2.01111111111111,11.1111111111111
2.4995,2.6,-2.00555555555556,11.1111111111111
2.5,2.6,-2,11.1111111111111
2.5,2.6,-2,11.1111111111111 
```
it now is
```csv
...
2.4985,-2.01666666666667,11.1111111111111
2.499,-2.01111111111111,11.1111111111111
2.4995,-2.00555555555556,11.1111111111111
2.5,-2,11.1111111111111
2.5,-2,0
```

@HansOlsson @GallLeo Please confirm.

hubertus65, 2018-11-22:
The difference in the MEdia result is ok and due to a bug fix.

HansOlsson, 2018-11-26:
The Test6 difference is due to whether time events are handled at the final point or not. Both variants are acceptable.

beutlich, 2018-11-26:
Thank you @tobolar @hubertus65  @HansOlsson

closed 2018-11-26 by beutlich