modelica/ModelicaStandardLibrary #2788: Have CLA assistant available for external contributors

beutlich, 2018-11-29:
@GallLeo We need to decide if and when we should setup CLA assistant (or similar service) for external contributors.

beutlich, 2018-11-29:
@hubertus65 @dietmarw This requires a clean markdown version of the CLA.

beutlich, 2019-03-26:
@GallLeo What is the current state regarding signed CLAs? Can we setup CLA assistant now?

beutlich, 2019-11-25:
It was decided among MAP-Lib members to install CLA Assistant by 2020-01-01 (though we probably will go for it in the second week of January).

@GallLeo It would be very helpful if we can figure out a list of all GitHub accounts that already signed the paper CLA (variants) by that time.

GallLeo, 2019-11-26:
We are currently doing two things:
1. Add Github IDs to the list of MA members (Google Spreadsheet)
2. Move the list of signed CLAs from Microsoft Excel to Google Spreadsheet

So, using/linking those two Google Spreadsheets, it should be trivial to extract the Github IDs which are covered by CLA.

beutlich, 2020-01-12:
* I installed the CLA Assistant using the [ModelicaAssociationCLA 1.1.1](https://github.com/modelica/ModelicaAssociationCLA/releases/tag/1.1.1) as [gist](https://gist.github.com/beutlich/0e7ff700866d7f52797145089c0eea9c).
* I also added all MAP-LIB members and MSL contributors that I know have signed the CLA.

PR #3306 (2020-01-13): Document CLA assistant for MSL contribution
Closes #2788.
closed 2020-01-13 by beutlich