modelica/ModelicaStandardLibrary #2789: Spice3 Capacitance documentation states capacitance can be negative

JoeRiel, 2018-11-30:
While a negative value may be handled by the engine, the Modelica type Modelica.SIunits.Capacitance has min=0, so using a negative value may result in a warning (which isn't unreasonable).

kristinmajetta, 2018-12-05:
So maybe the documentation should be changed.

beutlich, 2018-12-05:
> So maybe the documentation should be changed.

* or remove `min=0` from Modelica.SIunits.Capacitance
* or introduce a new Capacitance quantity type w/o the min restriction
* or simply accept the warning (which rather sounds like tool-specific beahviour)

beutlich, 2018-12-05:
Similarily. quantity type Mass also has `min=0`, but negative masses might be acceptable (to model holes).

kristinmajetta, 2018-12-05:
My favourite way would be the last suggested by Thomas Beutlich: "accept the warning". But I thought thats would not be possible concerning the rules of MSL?

HansOlsson, 2018-12-05:
Another alternative (I thought we had used it - but couldn't find any example) would be to modify the min-attribute for this parameter, so in Modelica.Electrical.Spice3.Basic.C_Capacitor have:
`parameter SI.Capacitance C(start=0, min=-Modelica.Constants.inf);`

However, considering that we have Resistance, Inductance and Capacitance (both as types and as somewhat similarly named models) I think they should be similar - and since the other two lack min-attribute I think this min-attribute could also be removed.

Note that the warning is really an error  - since the value should be consistent to the min-attribute according to the definition of "Real".

beutlich, 2018-12-05:
> so in Modelica.Electrical.Spice3.Basic.C_Capacitor have: `parameter SI.Capacitance C(start=0, min=-Modelica.Constants.inf):`

That is even possible (since the minimum attribute is not final) and sounds like the least intrusive fix. But we will likely miss some other similar cases.

> and since the other two lack min-attribute I think this min-attribute could also be removed.

Agreed. We should remove the minimum attribut for Capacitance and Mass generally.

Note, this is not for MSL v3.2.3, but the later one.

HansOlsson, 2018-12-05:
> Agreed. We should remove the minimum attribut for Capacitance and Mass generally.

For Mass it might be a more complicated - since there are so many uses of it where negative values would be a problem. I'm not fully against it, I just think we need to consider it more carefully.

JoeRiel, 2018-12-05:
The modification of min in the declaration in the Spice3 Capacitor seems the safest, particularly as the model itself doesn't really use the negative value other than as a flag to indicate that the user has not assigned a value.

PR #3232 (2019-11-19): Enable negative capacitance in SPICE3
As discussed, this pull request resolves #2789 locally in SPICE3 only.
closed 2019-11-19 by beutlich