modelica/ModelicaStandardLibrary #2792: Spice3 Constant Current source has no arrow

JoeRiel, 2018-12-01:
The symbol for Modelica.Electrical.Spice3.Sources.I_Current has no arrow, unlike the other current sources.  One has to use the solid/open ports to determine the prescribed flow direction, which is less clear and less convenient.

beutlich, 2018-12-02:
In fact, every icon looks slightly different w.r.t. to the blue circle and the position of the instance name.

beutlich, 2018-12-06:
And, while working on the Spice3.Sources, we should also fix the assignment of instance name and primary parameter:

[image]

PR #2797 (2018-12-06): Improve and fix icons of Modelica.Electrical.Spice3.Sources according to the icon convention
added arrow to symbol of Modelica.Electrical.Spice3.I_constant according to #2792

close #2792
commit 8a7b34e4 (2018-12-06): Merge pull request #2797 from kristinmajetta/master

Improve and fix icons of Modelica.Electrical.Spice3.Sources according to the icon convention (fixes #2792)
closed 2018-12-06 by dietmarw