modelica/ModelicaStandardLibrary #2802: Error in parameter comments in Buildings.Controls.OBC.CDL.Continuous.Hysteresis

milicag, 2018-12-07:
The comments that require fixing:

  parameter Real uLow(start=0) "if y=true and u<=uLow, switch to y=false";
  parameter Real uHigh(start=1) "if y=false and u>=uHigh, switch to y=true";

Suggested correction (please note the change to the inequality sign):

  parameter Real uLow(start=0) "if y=true and u<uLow, switch to y=false";
  parameter Real uHigh(start=1) "if y=false and u>uHigh, switch to y=true";

This change would be in line with the existing info section and the code implementation (y = not pre(y) and u > uHigh or pre(y) and u >= uLow).

<li> When the output was <b>false</b> and the input becomes
     <b>greater</b> than parameter <b>uHigh</b>, the output
     switches to <b>true</b>.</li>
<li> When the output was <b>true</b> and the input becomes
     <b>less</b> than parameter <b>uLow</b>, the output
     switches to <b>false</b>.</li>

dietmarw, 2018-12-07:
Looks like you got the wrong issue tracker. You probably want to report this here:
https://github.com/lbl-srg/modelica-buildings

milicag, 2018-12-08:
@dietmarw Sorry, there was an error in the title, I do want to report an error in the MSL hysteresis block.

PR #2803 (2019-02-07): Fix hysteresis comments
close #2802
closed 2019-02-07 by beutlich