modelica/ModelicaStandardLibrary #2808: Clarify removal of example model Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Inverter

beutlich, 2018-12-17:
Commit https://github.com/modelica/ModelicaStandardLibrary/commit/fa63e566a8c38e10a66a28b1f9e31565141d9e90 removed the example model Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Inverter where the commit message reads:

> Removing obsolete simulation model introduced by #2276 

However, Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_Inverter was not introduced by #2276, since it was added for MSL v3.2.2.

@christiankral Can you please confirm that the removal was on purpose?

christiankral, 2018-12-17:
Yes, the removal was performed on purpose, since the example is not really meaningful as a quasi static model. I certainly kept the transient model `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_Inverter`.

beutlich, 2018-12-17:
Thanks for clarification.

closed 2018-12-17 by beutlich