modelica/ModelicaStandardLibrary #2809: Analysis of the MSL v3.2.3-rc.2 regressions vs. v3.2.2+build.0-beta.3

beutlich, 2018-12-18:
This is the public [report](https://beutlich.github.io/MSL323FAILED-2/) of the 27 failed examples models for [MSL v3.2.3-rc.2](https://github.com/modelica/ModelicaStandardLibrary/releases/tag/v3.2.3-rc.2) vs. v3.2.2+build.0-beta.3 with the previous analysis results from #2783, leaving 20 models open for further analysis. Please note, that the 20 failing example models are no new regressions of MSL v3.2.3-rc.2, but simply were overlooked in the previous analysis #2783. :unamused:

### Modelica\Blocks\Examples\NoiseExamples\ActuatorWithNoise\ActuatorWithNoise.csv
### Modelica\Blocks\Examples\NoiseExamples\DrydenContinuousTurbulence\DrydenContinuousTurbulence.csv
### Modelica\Electrical\Analog\Examples\AmplifierWithOpAmpDetailed\AmplifierWithOpAmpDetailed.csv
### Modelica\Electrical\MultiPhase\Examples\Rectifier\Rectifier.csv
### Modelica\Fluid\Examples\Tanks\TanksWithOverflow\TanksWithOverflow.csv
### Modelica\StateGraph\Examples\ExecutionPaths\ExecutionPaths.csv
### Modelica\Utilities\Examples\ReadRealMatrixFromFile\ReadRealMatrixFromFile.csv

See #2783.

### Modelica\Electrical\Machines\Examples\AsynchronousInductionMachines\AIMC_Steinmetz\AIMC_Steinmetz.csv
### Modelica\Electrical\Machines\Examples\AsynchronousInductionMachines\AIMS_Start\AIMS_Start.csv
### Modelica\Electrical\Machines\Examples\SynchronousInductionMachines\SMEE_LoadDump\SMEE_LoadDump.csv
### Modelica\Electrical\Machines\Examples\SynchronousInductionMachines\SMPM_Braking\SMPM_Braking.csv
### Modelica\Electrical\Machines\Examples\Transformers\AIMC_Transformer\AIMC_Transformer.csv
### Modelica\Electrical\Machines\Examples\Transformers\AsymmetricalLoad\AsymmetricalLoad.csv
### Modelica\Electrical\Machines\Examples\Transformers\TransformerTestbench\TransformerTestbench.csv

@AHaumer @christiankral Please add you analysis here.

### Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMC_DOL\AIMC_DOL.csv
### Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMS_Start\AIMS_Start.csv
### Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMS_Start_MultiPhase\AIMS_Start_MultiPhase.csv
### Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\SMEE_Generator\SMEE_Generator.csv
### Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\SMEE_Generator_MultiPhase\SMEE_Generator_MultiPhase.csv

@AHaumer @christiankral Please add you analysis here.

### Modelica\Magnetic\QuasiStatic\FundamentalWave\Examples\BasicMachines\InductionMachines\IMC_DOL\IMC_DOL.csv
### Modelica\Magnetic\QuasiStatic\FundamentalWave\Examples\BasicMachines\SynchronousMachines\SMEE_Generator\SMEE_Generator.csv

@AHaumer @christiankral Please add you analysis here.

### Modelica\Mechanics\MultiBody\Examples\Elementary\DoublePendulumInitTip\DoublePendulumInitTip.csv
### Modelica\Mechanics\MultiBody\Examples\Elementary\PointGravityWithPointMasses2\PointGravityWithPointMasses2.csv
### Modelica\Mechanics\MultiBody\Examples\Loops\EngineV6\EngineV6.csv
### Modelica\Mechanics\MultiBody\Examples\Loops\EngineV6_analytic\EngineV6_analytic.csv
### Modelica\Mechanics\MultiBody\Examples\Rotational3DEffects\BevelGear1D\BevelGear1D.csv
### Modelica\Mechanics\MultiBody\Examples\Rotational3DEffects\GyroscopicEffects\GyroscopicEffects.csv

@tobolar Please add you analysis here.

AHaumer, 2018-12-21:
Modelica\Magnetic\QuasiStatic\FundamentalWave\Examples\BasicMachines\InductionMachines\IMC_DOL\IMC_DOL.csv
Modelica\Magnetic\QuasiStatic\FundamentalWave\Examples\BasicMachines\SynchronousMachines\SMEE_Generator\SMEE_Generator.csv

All of these: the differences are small, due to changed output interval (we unified that for all examples).
These are ok. New reference results (with new output interval length) have to be taken.

AHaumer, 2018-12-21:
Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMC_DOL\AIMC_DOL.csv
Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMS_Start\AIMS_Start.csv
Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\AIMS_Start_MultiPhase\AIMS_Start_MultiPhase.csv
Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\SMEE_Generator\SMEE_Generator.csv
Modelica\Magnetic\FundamentalWave\Examples\BasicMachines\SMEE_Generator_MultiPhase\SMEE_Generator_MultiPhase.csv

All of these: the differences are small, due to changed output interval (we unified that for all examples).
These are ok. New reference results (with new output interval length) have to be taken.

beutlich, 2019-01-04:
Modelica\Mechanics\MultiBody\Examples\Elementary\DoublePendulumInitTip\DoublePendulumInitTip.csv
This is due to #2568 and acceptable.

@tobolar Can you please analyze the other 5 MultiBody example models with regressions.

tobolar, 2019-01-10:
I've analysed the differences and they **all are acceptable**, see details below.

- Modelica.Mechanics.MultiBody.Examples.Elementary.**DoublePendulumInitTip**
  - Different start values, therefore offset in rotational angles; rotational velocities are identical

- Modelica.Mechanics.MultiBody.Examples.Elementary.**PointGravityWithPointMasses2**
  - Differences about 1e-13

- Modelica.Mechanics.MultiBody.Examples.Loops.**EngineV6**
  - Another model of gas force used; results ok and shall be taken as reference for further regression

- Modelica.Mechanics.MultiBody.Examples.Loops.**EngineV6_analytic**
  - ditto

- Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.**BevelGear1D**
  - Differences due to more precise definition of world.g in v3.2.3 (9.81 in v3.2.2 vs. 9.80665 m/s2).

- Modelica.Mechanics.MultiBody.Examples.Rotational3DEffects.**GyroscopicEffects**
  - ditto

beutlich, 2019-01-10:
Thanks, perfect.

closed 2019-01-10 by beutlich