modelica/ModelicaStandardLibrary #2813: Fix documentation of Translational User's Guide

christiankral, 2019-01-23:
Some documentation issue apply to:

- `Modelica.Mechanics.Translational.UsersGuide.Overview` 
- `Modelica.Mechanics.Translational.UsersGuide.SignConventions`
- `Modelica.Mechanics.Translational.UsersGuide.UserDefinedComponents`

The documentation text refers to:

- component `m1`
- component `m2`

Example: 
[image]

The depicted figures however show the components `mass1` and `mass2`. On my opinion this is misleading, as the shown component names of the figures and the text shall match. Otherwise it has to be explained what `m1` and `m2` are compared to `mass1` and `mass2`. This shall be fixed.

beutlich, 2019-01-24:
I removed the V: 3.2.3-beta.2 label since the issue also is present in earlier MSL versions and did not originate in V: 3.2.3-beta.2.

beutlich, 2019-01-24:
I changed the milestone for MSL4.0.0 since this is not a servere issue that needs back-porting for MSL 3.2.3.

PR #2867 (2019-07-16): Upgrade user's guide
For both **Rotational** and **Translational** libraries from Modelica.Mechanics.
This concerns the documentation text itself as well as proper html format (e.g. `&quot;` instead of `"`, etc.)

close #2813
closed 2019-07-16 by beutlich