modelica/ModelicaStandardLibrary #2814: Misleading doc string of force and torque sources

christiankral, 2019-01-23:
### Translational

In the model

- `Modelica.Mechanics.Translational.Sources.ConstantForce` 

the parameter 

```
parameter Modelica.SIunits.Force f_constant
    "Nominal force (if negative, force is acting as load)";
```

is used. This is somehow misleading as the negative force of the model only acts as load if the velocity is positive. For negative velocities, a negative force `f_constant` causes the model to generate power. Therefore, it is actually a matter of the quadrant in which the model operates (gray quadrants = not load). 
[image]
It thus makes sense to replace the doc string by, e.g.,  `"Nominal force (if negative, force is acting as load in positive direction of motion"`

In model `Modelica.Mechanics.Translational.Sources.SignForce` the same parameter with the same doc string is used, but the text is correct as the model always acts as load if `f_constant` is negative.

The models 

- `Modelica.Mechanics.Translational.Sources.LinearSpeedDependentForce`
- `Modelica.Mechanics.Translational.Sources.QuadraticSpeedDependentForce`

use the additional Boolean parameter `ForceDirection` to adapt the model behavior. Yet the negative sign of the force again refers to quadrant IV. Therefore the doc string might also be changed.

### Rotational

In model `Modelica.Mechanics.Rotational.Sources.ConstantTorque` the doc string of parameter `tau_constant` is correct: `"Constant torque (if negative, torque is acting as load in positive direction of rotation)"`. The same doc string is used for model `Modelica.Mechanics.Rotational.Sources.SignTorque`: in this case the "positive direction of rotation" does not apply, as it always behaves as load of the nominal torque is negative. The doc string `"Nominal torque (if negative, torque is acting as load)"` of the models

- `Modelica.Mechanics.Rotational.Sources.LinearSpeedDependentTorque`
- `Modelica.Mechanics.Rotational.Sources.QuadraticSpeedDependentTorque`

is not correct, however, as it again refers to quadrant IV only.

beutlich, 2019-01-24:
I changed the milestone for MSL4.0.0 since this is not a servere issue that needs back-porting for MSL 3.2.3.

tobolar, 2019-03-11:
Comparing the forces of the abovementioned translational components against speed:
[image]

commit 9bca2455 (2019-03-26): Fix description of force and torque sources (#2851)

* Fix #2814

* Refs #2814 and #2853: Changed 'movement' to 'motion'

in parameter ForceDirection
commit 20597c89 (2019-05-28): Fix description of force and torque sources (#2853)

* Fix #2814

* Refs #2814 and #2853: Changed 'movement' to 'motion'

in parameter ForceDirection
closed 2019-03-26 by beutlich