modelica/ModelicaStandardLibrary #2815: moist air at very low temperatures

thorade, 2019-02-05:
The nonlinear solver sometimes uses very low or even negative temperatures in function calls to `h_pTX`, resulting in errors, because of the `log(T)` in the `h_Tlow` function. 
I suggest to catch negative temperatures and return h=-inf for T<=0, will send a PR.

The following model was used to investigate and it revealed another problem:
```model MoistAirTLowTest
  replaceable package Medium = Modelica.Media.Air.MoistAir;
  Modelica.Blocks.Sources.Ramp Tramp(
    height=-320,
    duration=0.8,
    offset=323.15,
    startTime=0.1)
    annotation (Placement(transformation(extent={{-80,38},{-60,60}})));

  Medium.AbsolutePressure pamb= 101325;
  Medium.Temperature T;
  Medium.SpecificEnthalpy h;

equation 
  T = Tramp.y;
  h = Medium.h_pTX(pamb,T,Medium.reference_X);
  annotation (uses(Modelica(version="3.2.2")));
end MoistAirTLowTest;
```
After simulation, plot h with T as independent variable. The bump in the curve might push the solver even further in the wrong direction, I believe. 

[image]

dietmarw, 2019-02-05:
Next time, no need to create the issue. Just send the PR :-)

beutlich, 2019-02-05:
> 
> 
> Next time, no need to create the issue. Just send the PR :-)

Did I miss something?

dietmarw, 2019-02-05:
The PR is basically an issue in itself. If somebody like @thorade is so nice to offer a PR they can simply save the effort of creating a separate issue.

beutlich, 2019-02-05:
> will send a PR.

Ah, I missed it.

thorade, 2019-02-05:
The PR will only address the safeguard for log(T) but not the little bump.
Ah, and forgive the crappy commit message, only thought about it after pushing

beutlich, 2019-02-05:
> Ah, and forgive the crappy commit message, only thought about it after pushing

You still can git commit -ammend and git push --force, but otherwise, if we squash merge ...