modelica/ModelicaStandardLibrary #2820: Update ModelicaReference to latest Modelica specification 3.4

beutlich, 2019-02-06:
ModelicaReference is based on MLS 3.2 (not even sure which revision), but should be updated to MLS 3.4.

HansOlsson, 2019-02-14:
Makes sense, will see when I get time, and I assume it is best to ensure that the version on maint/3.2.3 uses MLS 3.2r2 as the rest. 
I realized that we should also update https://github.com/modelica/Modelica-Compliance to MLS 3.4; and it may make sense to update both together (in the sense that changes are both described and tested).

beutlich, 2019-08-09:
Seehttps://htmlpreview.github.io/?https://github.com/modelica-tools/ModelicaSyntaxChecker/blob/master/Grammar.html for the grammar.

beutlich, 2019-09-26:
@dietmarw @HansOlsson Do you think you can manage to update the ModelicaReference?

dietmarw, 2019-09-27:
@svorkoetter Could you provide the updated "Modelica Grammar" also including fixes to HTML issues that were mentioned earlier. Feel free to submit it as PR to https://github.com/modelica-tools/ModelicaSyntaxChecker (including updated moparser binary if you got them). Thanks.

dietmarw, 2019-09-27:
@HansOlsson since you are working most intense with the Specs I wonder if you can list the new/changed parts from Spec 3.2 to 3.4 that are relevant for Modelica Refernece. Otherwise I need to go through the documents with revision marks and chances are I might miss some central items/changes.

HansOlsson, 2019-09-27:
I will try to do it, I just started with looking at Modelica-Compliance (well, after looking at issues for the specification itself).

svorkoetter, 2019-09-27:
Will do. Just so I'm sure to get the right grammar, what's the official source of the up-to-date spec?

HansOlsson, 2019-09-27:
> Will do. Just so I'm sure to get the right grammar, what's the official source of the up-to-date spec?

Do you mean https://github.com/modelica/ModelicaSpecification
which has links to PDF and HTML-versions, both for master-branch and 3.4.

svorkoetter, 2019-09-27:
@HansOlsson, has the grammar changed since the time that I already uploaded the requested files?

beutlich, 2019-10-10:
@svorkoetter Can you please create a PR updating the grammar to specification 3.4. See https://specification.modelica.org/v3.4/A2.html#grammar for the reference to take it from.

beutlich, 2019-10-10:
> has the grammar changed since the time that I already uploaded the requested files?

Yes, it did. Take the pure/impure class-prefixes as example.

beutlich, 2019-10-10:
@HansOlsson Should the clocked/synchronous language elements also be added to ModelicaReference?

HansOlsson, 2019-10-11:
> @HansOlsson Should the clocked/synchronous language elements also be added to ModelicaReference?

I believe so, especially since we are adding Modelica.Clocked. 
Note that the state-machine part can be skipped at the moment.

Adding sample, hold etc in the list of operators would be straightforward; but it is not clear if we should add a general overview - and if so how detailed. If we add it my preference would be to keep it very short - so just a one (or two) paragraph summary, since it is complicated to maintain ModelicaReference as a duplicated variant of the specification.

beutlich, 2019-10-11:
> If we add it my preference would be to keep it very short

Yes, please!

svorkoetter, 2019-10-11:
> Yes, it did. Take the pure/impure class-prefixes for example.

That was already there. The only changes seem to be the missing closing `</p>` tags, and the change from `annotation` to `annotation_comment`.

beutlich, 2019-11-04:
The syntax of the keywords, e.g. in [ModelicaReference.'import'](https://doc.modelica.org/ModelicaReference%203.2.3/Resources/helpDymola/ModelicaReference.html#ModelicaReference.'import') is

* a partial duplication of the [Modelica Grammar](https://doc.modelica.org/ModelicaReference%203.2.3/Resources/helpDymola/ModelicaReference.html#ModelicaReference.ModelicaGrammar)
* outdated, since it still is for Modelica language 3.2

Can we savely remove it to avoid the extra maintenance and link to ModelicaReference.ModelicaGrammar instead? Unfortunetaly, it currently is not possible to refer to anchor links within modelica URIs, see https://github.com/modelica/ModelicaSpecification/issues/2445 for the corresponding specification issue.

dietmarw, 2019-11-05:
I agree. Instead of listing the syntax we should simply point to `ModelicaRerence.ModelicaGrammar`.  In a addition the example should be changed since it demonstrates the usage of wildcard imports, something that we at the same time emphasise to not use.

PR #3443 (2020-02-25): Add synchronous language elements to ModelicaReference
Add something for synchronous ans state machines.
The main idea was to add operators so that they can be found.

Resolves #2820
closed 2020-02-25 by beutlich