modelica/ModelicaStandardLibrary #2832: Clean up release page?

HansOlsson, 2019-02-14:
The main page links to: https://github.com/modelica/ModelicaStandardLibrary/releases

That seems useful to access older released versions when needed, but would it be possible to not list (or in some other hide) pre-release versions superseded by other releases (both actual releases and later pre-releases)? That would make it easier to find the relevant ones.

The "tags"-tab can be left as it is.

(Assuming I don't have any weird settings causing this.)

dietmarw, 2019-02-14:
I guess simply removing the pre-releases once the final release is done should be safe to do. As @HansOlsson says, the tags will still be there.

beutlich, 2019-02-18:
I agree with you desire to exclude pre-releases from visibility. That should be a general GitHub option in my opinion. I will not touch the pre-releases as they are the only source for downloading the appropriate assets.

closed 2019-02-18 by beutlich

HansOlsson, 2019-02-18:
> That should be a general GitHub option in my opinion.

I agree that this should be ideal, but I haven't found it. 
So, I will just open a support request.

> I will not touch the pre-releases as they are the only source for downloading the appropriate assets.

Ok, that makes sense. I noticed that they are also found clicking on the "notes" on the tags - but I assume that is just the internal format.