modelica/ModelicaStandardLibrary #2836: Update PRNGs in ModelicaRandom.c?

beutlich, 2019-02-21:
When ModelicaRandom.c was introduced with the Noise package for MSL v3.2.2 the **xorshift** PRNGs from http://xoshiro.di.unimi.it/xorshift.php were utilized. These PRNGs are now superseded by **xoshiro**, which according to the authors at http://xoshiro.di.unimi.it are faster and better.

@MartinOtter @tobolar @HansOlsson What do you think?

HansOlsson, 2019-02-21:
It seems like an improvement. 

But I don't think increasing the performance and those statistical properties is high priority.

A concern for _any_ change of PRNG is that we provide the possibility to select a seed to be able to reproduce simulations - and changing PRNG will break that. Hopefully it is sufficient to just document that it has changed - and anyone really concerned about reproducing simulation will have a copy of the old library.

beutlich, 2019-02-21:
> It seems like an improvement.

Exactly.

> A concern for _any_ change of PRNG is that we provide the possibility to select a seed to be able to reproduce simulations - and changing PRNG will break that. Hopefully it is sufficient to just [...]

Note that compatibility already is broken (and documented) in v3.2.3 w.r.t. v3.2.2 due to #2252.

tobolar, 2019-02-25:
Modelica counterparts of the C implementation are `Modelica.Math.Random.Generators.Xorshift...`. Changing PRNGs will require to rename those classes as well or to add new ones. So the user will be clearly informed about the change.
... and introducing MSL v4 will make it even easier (I guess) to get rid of xorshift-based generators.

MartinOtter, 2019-04-25:
You are right, xoshiro should be used at some point in the future. However, I would wait and not do it now, because making this change in a correct way might be difficile and requires thorough testing (e.g. as I recall, one low order random number generator is used to initialize a high order random generator and one would have to dig in all these codes to figure out that all this still works with the improved generators).