modelica/ModelicaStandardLibrary #2845: Should Modelica.StateGraph be obsolete in 3.2.3?

HansOlsson, 2019-03-05:
In 3.2.3 the entire Modelica.StateGraph was marked as obsolete without extending from ObsoleteModel; and without having a clear replacement.

Having it obsolete seems a bit excessive - I thought we wanted to merge Modelica_StateGraph2 improvements into Modelica.StateGraph (and possible move Modelica.StateGraph).

Note: Modelica.StateGraph.Temporary is both marked as obsolete and extending from ObsoleteModel.

HansOlsson, 2019-03-05:
Ah, mistake from me.

Note: Modelica.StateGraph.Temporary is extending from ObsoleteModel; but not marked as obsolete.

I believe this makes it even clearer that it is just a typo.

beutlich, 2019-03-05:
Confirmed, 94143015905d8bd7dddb7c97e55a576f57001060 added the obsolete annotation on the wrong package.

PR #2847 (2019-03-06): Fix obsolete annotation
Fix #2845 (in maint/3.2.3branch).
closed 2019-03-06 by beutlich