modelica/ModelicaStandardLibrary #2856: Modelica.Media.Incompressible documentation question

lyebra, 2019-03-19:
In the docu the specific enthalpy definition seems to be written with a typo error in the sign, that is positive (h = u + p/d).

beutlich, 2019-03-19:
Do you mean https://github.com/modelica/ModelicaStandardLibrary/blob/ec70d1f10887371f11f14cb1abb696b60ee8cd6f/Modelica/Media/Incompressible.mo#L843?

lyebra, 2019-03-19:
Yes, the typo error is in the minus sign in h = u - p/d. Actually it is 
h = u + p/d.

Best regards,

Luis

-- 
--------------------------------------------------------------------------
  Dr. Luis José Yebra Muñoz
  CIEMAT - Plataforma Solar de Almería

  Crta. de Senés, s/n
  04200  TABERNAS
  ALMERÍA

  Tlf: +34 950 387800. Ext.:923
  Fax: +34 950 365015
  Web: www.psa.es

  E-mail: luis.yebra@psa.es
--------------------------------------------------------------------------

beutlich, 2019-03-19:
@thorade Do you confirm?

HansOlsson, 2019-03-19:
I agree that the minus sign is either a simply typo or someone intending to write "u=h-p/d".

But to me the conclusion doesn't really follow in this simple way: just because h=u+p/d doesn't mean that h necessarily depend on p. It could be that h is independent of p and it is the specific entropy u=h-p/d that must depend on pressure.

lyebra, 2019-03-19:
Hi, All;

Hans, it is my pleasure to meet you again (by e-mail)...

Just a small comment by my side. The specific enthalpy is defined in the 
form h=u+p/d. I understand that in Media library we can approximate the 
dependence of the enthalpy with the pressure for incompressible media as 
explained in the text in the docu. I think that it would be enough, and 
convenient, just to change the sign.

Best!

    Luis

HansOlsson, 2019-03-19:
My point was that the conclusion only follows if u is necessarily independent of p.

If we look at the flags we see that enthalpyOfT decides whether h depends on T or p, T.
And singleState decides whether we use u=h-p/d or u=h-p0/d. 

By setting enthalphyOfT=true and singleState=false we have h independent of p and u dependent on p - i.e. exactly the case where the conclusion doesn't hold.

Obviously a more interesting question is how a real media behaves. If we modify Modelica.Media.Examples.SimpleLiquidWater to use StandardWater instead of simplified water we see that with increasing p, then h increases by about 8.5 and u decreases by about 1.6; at 330 K; whereas at 280 K then the slopes are 9.9 and -0.1. I don't know how representative this is.
 
I believe that a correct explanation is that 

However, enthalpy and inner energy can not both be independent of pressure since u = h - p/d. For liquids it is anyway common to neglect this dependence for both of them since for constant density the neglected term is (p - p0)/d, which in comparison with cp is very small for most liquids.

thorade, 2019-03-19:
For real fluids, all state properties are connected to each other, and you can only have two independent variables for a single phase. 
For liquids (almost incompressible), and with p,T as independent variables, enthalpy changes very little with pressure, and almost linear with temperature (with heat capacity as factor). 
In other words: (dh/dp)_T << (dh/dT)_p.

After reading that section of the documentation several times, I would say it is not wrong, except the sign in the formula. Someone trying to really understand fluid properties should probably read a thermodynamics book.

The easy and immediate fix is to just correct the sign, and I will submit a PR in a couple of minutes.

closed 2019-03-20 by beutlich

casella, 2019-05-17:
> However, enthalpy and inner energy can not both be independent of pressure since u = h - p/d.

For the record, this statement is not true. Consider ideal gases, where p/d = R*T, so that both h = h(T) and u = u(T) without any approximation.

However, this doesn't change the conclusions of the ticket.

PR #2859 (2019-03-20): Fix typo in enthalpy formula
this closes #2856