modelica/ModelicaStandardLibrary #2860: Modelica.Blocks.MathBoolean.Nand has wrong documentation for empty input

HansOlsson, 2019-03-20:
When looking at #682 I found that Nand-block documentation says that no inputs gives the output "false" - but the result is "true". It should likely be corrected together with #682.

beutlich, 2019-04-09:
With #2888 applied, it is MathBoolean.And that needs to be updated, while Nand will be correct then:
```mo
  block And "Logical 'and': y = u[1] and u[2] and ... and u[nu]"
    extends Modelica.Blocks.Interfaces.PartialBooleanMISO;

  equation
    y = Modelica.Math.BooleanVectors.allTrue(
                              u);
    annotation (defaultComponentName="and1", Icon(graphics={Text(
              extent={{-76,40},{60,-40}},
              textString="and")}),
      Documentation(info="<html>
<p>
The output is <strong>true</strong> if all inputs are <strong>true</strong>, otherwise
the output is <strong>false</strong>.
</p>

<p>
The input connector is a vector of Boolean input signals.
When a connection line is drawn, the dimension of the input
vector is enlarged by one and the connection is automatically
connected to this new free index (thanks to the
connectorSizing annotation).
</p>

<p>
The usage is demonstrated, e.g., in example
<a href=\"modelica://Modelica.Blocks.Examples.BooleanNetwork1\">Modelica.Blocks.Examples.BooleanNetwork1</a>.
</p>

<p>
If no connection to the input connector \"u\" is present,
the output is set to <strong>true</strong>: y=true.
</p>
</html>"));
  end And;
```

beutlich, 2019-06-13:
Resolved in master by #2916: The behaviour of And and Nand block was changed to return true on empty on non-connected input connector, respectively false. In that sense, it no longer is a pure documentation issue.

We could argue, that we might want to add All and Nall blocks rstoring the existing beahviour (and with conversions And->All and Nand->Nall applied to not break compatibility).

@HansOlsson @MartinOtter Feel free to reopen if you prefer any other implementation.

closed 2019-06-13 by beutlich