modelica/ModelicaStandardLibrary #2862: Integrate Modelica_Synchronous into MSL 4.0.0

bernhard-thiele, 2019-03-20:
Add the [Modelica_Synchronous](https://github.com/modelica/Modelica_Synchronous) library to the MSL 4.0.0.

Some thoughts how the integrated structure could look like:

```
Modelica
  .ClockedBlocks                 <- Modelica_Synchronous (M_S)
-> Pro: Self-contained library, consistent with approach taken for .ComplexBlocks
-> Con: Additional first level library which could be subsumed under the broader domain of directed data-flow blocks

Modelica.Blocks
  .Clocked                       <- M_S
-> Pro: Subpackage below an existing adequate domain, no need for additional first level Modelica.xxx library
-> Con: Basically adds a self-contained sub-library into a self-contained library which feels not coherent

Modelica.Blocks
  .Examples.Clocked              <- M_S.Examples
  .Clocks
    .Sources                     <- M_S.ClockSignals.Clocks
    .Sampler                     <- M_S.ClockSignals.Sampler
    .Interfaces                  <- M_S.ClockSignals.Interfaces
  .ClockedReal                   <- M_S.RealSignals
  .ClockedInteger                <- M_S.IntegerSignals
  .ClockedBoolean                <- M_S.BooleanSignals
  .Types                         <- M_S.Types
-> Pro: Arguably more coherent integration into the Blocks library
-> Con: Four additional subpackages below Blocks, not as self-contained in regard to maintenance responsibilities

Alternative (possibly not as good?) subpackage names instead of Modelica.ClockedBlocks
Modelica
  .Synchronous
  .ClockedSynchronous
```

bernhard-thiele, 2019-03-20:
@HansOlsson @MartinOtter @christoff-buerger @tbeu @tobolar  Just mentioning you so that you are aware that I have created a ticket for discussing the Modelica_Synchronous integration

HansOlsson, 2019-03-25:
I think I prefer the Modelica.Blocks.Clocked

Having it split just adds a complication (finding all of them together - and accidentaly mixing with non-clocked), and I think we already have too packages directly under Modelica.

beutlich, 2019-03-25:
I have no strong opinion here if we should aim for first or second alternative here. But since we have ComplexBlocks already next to Blocks (and ComplexMath next to Math), and also Modelica_DeviceDrivers has ClockecBocks next to Blocks, it might look naturals to add ClockedBlocks too. It also seems to feature least effort for migration and a potential later modularization of domains.

christoff-buerger, 2019-04-03:
I think I would prefer `Modelica.Synchronous`.

Are there specific reasons to use `ClockedBlocks` -- to talk about blocks? I somehow miss the context with blocks here. Of course, it is very likely that in most use-cases sampled subsystems are simple block diagrams. But for me this tastes to much Simulink like. I mean, there is no need to only use blocks within clocked partitions. You can use any Modelica model. For me this is actually a strong selling point of Modelica Synchronous (not Modelica Clocked Blocks), that you can apply it on existing continuous models. After all, clocks can have solvers for _clocked discretized continuous-time partitions_; this does not sound like just a block-diagram thing to me.

Or is the use of block due to synchronous elements (like sample, hold, clock) being blocks? But also that are very strange/special blocks for me, not comparable with normal blocks. The catch with clock partitioning _is_, that it is cross-cutting -- i.e., breaking -- the normal component hierarchy; but a data-flow according to a component hierarchy _is_ the essential ingredient for blocks.

I hope this comment is not to confusing; I have trouble to explain my bad feeling about relating synchronous to blocks concise.

bernhard-thiele, 2019-04-04:
At the moment all drag'n'drop components from the library are blocks and use the directed data-flow connectors from `Modelica.Blocks.Interfaces` and/or the directed Clock connectors. From that perspective I think it is okay to classify them as blocks, but I kind of get your preference for `Modelica.Synchronous`.

What I don't like  with the name `Synchronous` is that we have the "synchronous data flow principle" as a fundamental concept of Modelica (MLS 8.4), so all of Modelica is basically "synchronous". For a more unambiguous classification, one could use the name `Modelica.ClockedSynchronous`, but this has the disadvantage of being less short and crisp.

MartinOtter, 2019-04-25:
I am strongly to include it in parallel to Blocks due to practical reasons (and since Modelica.Synchronous are discrete-time blocks, whereas Modelica.Blocks are continouous-time blocks): Subpackages are structured into Examples and Interfaces and it would be very awkward if the many examples and interfaces in Modelica_Synchronous would be spread in many locations of Modelica.Blocks. Including everything (examples, several Interface packages) under Modelica.Blocks.Synchronous would suddenly introduce a new structure.

Furthermore, as Christoff pointed out, there are clocked continuous-time models used in side a sampled block and this looks strange for a Modelica.Blocks library (so it should not be in Modelica.Blocks).

Modelica has synchronous language elements to describe clocked state machines. If the package is called Modelica.Synchronous, it would be natural to include clocked state machines also here in the future (just point this out, although this is not yet relevant).

Here are my preferences (first: my preferred choice):

1. Modelica.Clocked
2. Modelica.Sampled
3. Modelica.Synchronous

HansOlsson, 2019-04-25:
Modelica.Sampled would be confusing for many as the non-clocked sample-operator also leads to sampled systems.

> What I don't like with the name Synchronous is that we have the "synchronous data flow principle" as a fundamental concept of Modelica (MLS 8.4), so all of Modelica is basically "synchronous".

I know that we have stated that for a long time, and synchronous data flow was part of the original inspiration for Modelica (as described in "Hybrid Modeling in Modelica based on the Synchronous Data Flow Principle"). However, looking at https://en.wikipedia.org/wiki/Dataflow_architecture the general data flow terminology doesn't seem like a good match for the entire Modelica language; and people don't normally use this term for Modelica.

However, that doesn't imply that I like the name Modelica.Synchronous.

MartinOtter, 2019-04-30:
> Modelica.Sampled would be confusing for many as the non-clocked sample-operator also leads to sampled systems.

What about **Modelica.Clocked**: Isnt't this exactly describing the core feature? Yes, there is also a block Modelica.Blocks.Sources.Clock, but I think it is fine to have a continuous-time clock and call it **Clock**. **Clocked** on the other hand means that the models/blocks inside this package are executed under control of a clock.

>> What I don't like with the name Synchronous is that we have the "synchronous data flow principle" as a fundamental concept of Modelica (MLS 8.4), so all of Modelica is basically "synchronous"

I think the underlying principle of Modelica is usually stated as **single assignment rule**: a variable is assigned exactly once.

> I know that we have stated that for a long time, and synchronous data flow was part of the original inspiration for Modelica (as described in "Hybrid Modeling in Modelica based on the Synchronous [...]

I would also like to mention https://en.wikipedia.org/wiki/Synchronous_programming_language that describes **synchronous programming languages**. These languages use **single assignment** at a particular **clock tick** (a variable can be assigned in multiple statements, but at the same clock tick it can be assigned only once). 

Therefore, I still think that **Modelica.Synchronous** would be o.k., but I prefer **Modelica.Clocked** because people who do not know synchronous languages have an intuitive feeling for "Clocked" but "Synchronous" has several meanings and it is not at once obvious what meaning is meant at this place.

HansOlsson, 2019-04-30:
> What about Modelica.Clocked: Isnt't this exactly describing the core feature?

I have no problem with Modelica.Clocked. 
However, I prefer Modelica.Blocks.Clocked since I think we are adding too many packages directly under Modelica, but it might be that the problem isn't Clocked but some of the other packages (e.g. ComplexMath and ComplexBlock).

> I think the underlying principle of Modelica is usually stated as single assignment rule: a variable is assigned exactly once.
> I would also like to mention https://en.wikipedia.org/wiki/Synchronous_programming_language that describes synchronous programming languages. 

I opened a separate ticket on that https://github.com/modelica/ModelicaSpecification/issues/2348

bernhard-thiele, 2019-04-30:
From the discussion so far there seems a majority preference to have it as a top level package, despite the drawback of increasing the package count below Modelica.

*Modelica.Sampled* fits, but as Hans mentioned, it might be more confusing than the other options.

**Modelica.Clocked** is short, intuitive, less ambiguous for the the uninitiated and I like it, too.

I have no strong feelings against **Modelica.Synchronous** either. It is in line with the related MLS chapter name and has the advantage that it is the direct mapping of the current library name into the MSL namespace, so that at least people who know **Modelica_Synchronous** will immediately recognize what **Modelica.Synchronous** is about.

If we would open up the package for state machines at a later time, I think both names would do.

christoff-buerger, 2019-05-07:
I like both, `Modelica.Clocked` and `Modelica.Synchronous`.

I slightly prefer `Modelica.Synchronous` simply for the reason that we speak of synchronous language features since their introduction in the specification quite some time ago. Thus, the papers documenting the foundations and how to use clocks etc use the term _synchronous_; and the specification speaks of _Synchronous Language Elements_ in Chapter 16, not clocked. But maybe it would be better (in the long run) to consistently change it all to clocked? If not, I prefer `Modelica.Synchronous`.

**Note:** A short scan of the specification shows, that synchronous is much less used than clocked in Chapter 16 _Synchronous Language Elements_ and, as mentioned, overlaps with Section 8.4 _Synchronous Data-flow Principle and Single Assignment Rule_.

PR #2949 (2019-12-02): Integrate the Modelica_Synchronous library into the MSL 4.0.0
**Goal:** Integrate the `Modelica_Synchronous` library into the MSL 4.0.0 under the name `Modelica.Clocked` (closes #2862).

This PR contains the most important adjustments for integrating the Modelica_Synchronous library into the MSL. So hopefully, remaining work is mostly fine tuning. 

After a very good discussion in #2862, I decided to integrate the library under the name `Modelica.Clocked`, hoping that it is a good compromise given the various pros and cons.
closed 2019-12-02 by beutlich