modelica/ModelicaStandardLibrary #2866: Axis for ModelicaTest.MultiBody.Joints.UniversalSpherical

HansOlsson, 2019-03-27:
Based on https://github.com/modelica/ModelicaStandardLibrary/issues/2720#issuecomment-477157967
as far as I can tell there are three variants of universal+rod+spherical, and it is checked they generate the same result.

However, the middle one seems to have strange axis, a different n1 that is co-linear with the rod-axis. I would propose to change it to {0,0,1} as the other two variants and change start-value back to zero.

MartinOtter, 2019-04-25:
There is no "n1" defined. Can you give a more detailed description based on oject names (e.g. universal.n_a should be changed from xxx to yyy).

HansOlsson, 2019-04-25:
> There is no "n1" defined and your statement is not clear.

That was correct in the linked comment:
https://github.com/modelica/ModelicaStandardLibrary/issues/2720#issuecomment-477080461

I meant that both `universalSpherical.n1_a={0,0,1}` and `universalSpherical1.n1_a={0,0,1}` but `universal.n_a={1,0,0}` which is both different and co-linear with `fixedTranslation.r`.

Thus I propose to change `universal.n_a={0,0,1}`, and reverting the change of `universal.phi_a `back to zero.

If this isn't correct there should be some additional comments; because it clearly seems that the intent is to ensure that three different variants give the same result.

MartinOtter, 2019-04-25:
Yes, you are right (so to change to universal.n_a={0,0,1}, and reverting the change of universal.phi_a back to zero).

MartinOtter, 2019-04-25:
@tobolar: Please, can you perform this change.

tobolar, 2019-04-29:
Will do it.

PR #2926 (2019-05-20): Proper axes orientation and initialization of UniversalSpherical
close #2866
closed 2019-05-20 by beutlich