modelica/ModelicaStandardLibrary #2868: Modelica.Media.Air.MoistAir.density_derX is not right.

finback-at, 2019-03-29:
I think that the current algorithms of Modelica.Media.Air.MoistAir.density_derX,
```
dddX[Water] := pressure(state)*(steam.R - dryair.R)/((steam.R - dryair.R)
  *state.X[Water]*temperature(state) + dryair.R*temperature(state))^2;
dddX[Air] := pressure(state)*(dryair.R - steam.R)/((dryair.R - steam.R)*
  state.X[Air]*temperature(state) + steam.R*temperature(state))^2;
```
are not right.  

**Unit check**

Mass fraction X [kg/kg] is unitless.
The unit of density_derX should be same as density.
But the unit of abouve equations are,
`pressure*R/(R*temperature)^2 = pressure/(R*temperature^2) = density/temperature.
`
     where, `density = pressure/(R*temperature)`   is equation of state for ideal gases.

 **my derivation**
```
density = p/(R*T)
=>
d(density)/dX = - p/(R^2*T) * dR/dX = {-p*(dR/dX)} / (R^2*T)  --- (a)
```
(1) dddX[Water]

```
R = steam.R*X[Water] + dryair.R*(1-X[Water]) 
   = (steam.R - dryair.R)*X[Water] + dryair.R 

dR/dX[Water] = steam.R - dryair.R

numerator of (a) = -p*(dR/dX) = - pressure(state)*(steam.R - dryair.R)
denominator of (a) = R^2*T 
   = ((steam.R - dryair.R)*X[Water] + dryair.R)^2*temperature(state)

dddX[Water] = - pressure(state)*(steam.R - dryair.R)/
    ( ((steam.R - dryair.R)*X[Water] + dryair.R)^2 * temperature(state))
```

(2) dddX[Air]
```
R = steam.R*(1-X[Air]) + dryair.R*X[Air]
  = steam.R + (dryair.R - steam.R)*X[Air]

dR/dX[Air] = dryair.R - steam.R

numerator of (*) = -p*(dR/dX) = - pressure(state)*(dryair.R - steam.R)
denominator of (*) = R^2*T
   = (steam.R + (dryair.R - steam.R)*X[Air])^2 * temperature(state)

dddX[Air] = - pressure(state)*(dryair.R - steam.R)/
              ( (steam.R + (dryair.R - steam.R)*X[Air])^2 * temperature(state) )
```

beutlich, 2019-03-29:
That code is in there since MoMoLib was integrated more than 7 years ago: https://github.com/modelica/ModelicaStandardLibrary/blob/ba2f041349fe1333d2124bef5ccdcf68a033faa3/Modelica/Media/Air.mo#L3734-L3745

@wischhusen  Can you please have a look at this.

@HansOlsson You once told (maybe for a similar fix of molarMass), that unit checks in function are tricky. Can you please cross-check.

wischhusen, 2019-03-29:
Looks reasonable. Have to revisit the code later.

PR #2932 (2019-07-24): Fix Media.Air.MoistAir.{density_derX, s_pTX, s_pTX_der}
Corrected issues #2874 #2872 #2868 as elaborated by the reporter.

Fix #2868 (density_derX), fix #2872 (s_pTX) and fix #2874 (s_pTX_der).
closed 2019-07-24 by beutlich