modelica/ModelicaStandardLibrary #2870: algorithm of Modelica.Media.IdealGases.Common.Functions.s0_Tlow_der

finback-at, 2019-03-29:
Current algorithm of s0_Tlow_der is 
```
  s := data.R*(data.blow[2] - 0.5*data.alow[
    1]/(T*T) - data.alow[2]/T + data.alow[3]*Math.log(T) + T*(
    data.alow[4] + T*(0.5*data.alow[5] + T*(1/3*data.alow[6] + 0.25*data.alow[
    7]*T))));
```
It is just as same as s0_Tlow.

From the relation 
   (ds/dT)_p = Cp/T,
I think it should be changed to 
```
s := T_der * Modelica.Media.IdealGases.Common.Functions.cp_Tlow(data,T)/T;
``

beutlich, 2019-03-29:
@thorade Even though I assigned to @hubertus65 can you please have a look at it.

thorade, 2019-03-30:
Will have a look next week when I am back from travelling. 
Obviously, any property and its derivative cannot have the identical algorithm.

beutlich, 2019-05-03:
> 
> 
> Will have a look next week when I am back from travelling.

Did you find some time?

thorade, 2019-05-05:
Sorry for answering so late. 

@finback-at you can create permanent links to code snippets:
https://help.github.com/en/articles/creating-a-permanent-link-to-a-code-snippet
Makes it a few seconds faster to find the code and 100% unambigous

So this is referring to these lines:
https://github.com/modelica/ModelicaStandardLibrary/blob/30e08efd288786e34a2adabd64e50a38a5ac7490/Modelica/Media/IdealGases/Common/Functions.mo#L169-L194

First part: Yes, the code is identical, and this cannot be correct. Only difference is the additional but unused T_der input.

Second part: entropy and also cp depend on T only in this model, so the general approach is good (and also used for enthalpy derivative). As cp depends on T also, the formula is incorrect, compare 
https://www.wolframalpha.com/input/?i=1%2Fx*f(x)+derivative

@beutlich Do we have tests for media derivatives somewhere? Probably not for this one, but we should have one.

@finback-at Do you know how to create Pull Requests?
https://guides.github.com/activities/forking/

beutlich, 2019-05-06:
> @beutlich Do we have tests for media derivatives somewhere? Probably not for this one, but we should have one.

Probably not. Could you invent one test model?

finback-at, 2019-05-07:
Originally, s0(T) is obtained from
https://www.grc.nasa.gov/WWW/CEAWeb/TP-2002-211556.pdf, p.2, eq.(3).
It is written  below eq.(1) of this document,
"Enthalpy and entropy are obtained by integrating Cpo (T ) and Cpo(T )/ T , respectively, with respect to T:",
 and it is easy to check the equation by hand.

I think it is simple to write,
  s0_Tlow_der = Cp_Tlow(data,T) / T,
just like as,
  h_Tlow_der = Cp_Tlow(data.T),
in current codes.

finback-at, 2019-05-07:
I checked the code in Fix algorithm of s0_Tlow_der #2919 is same as Cp_Tlow(data.T)/T.
Thank you!.
Thank you to let me know about how to create pull request. I will try.

finback-at, 2019-05-07:
Sorry, I dropped T_der in may comments

s0_Tlow_der =T_der* Cp_Tlow(data,T) / T,
h_Tlow_der = dT*Cp_Tlow(data.T),
And I checked the Fix algorithm is same as T_der*Cp_Tlow(data,T)/T.

beutlich, 2019-05-07:
> Thank you to let me know about how to create pull request. I will try.

Already done by #2919.

thorade, 2019-05-07:
This seems to be related to #1909 also.
As s0_Tlow uses smooth annotation and not derivaitve annotation, tools will differentiate themselves as required.
The only place I could find where this function is actually called is from here:
https://github.com/modelica/ModelicaStandardLibrary/blob/d338e4d603f2f5872f0f32098ae68adbcec41191/Modelica/Media/Air/MoistAir.mo#L1409

thorade, 2019-05-07:
@phjordan Isn't this the function you investigated two years ago also? 
What was your conclusion back then?

thorade, 2019-05-09:
> > @beutlich Do we have tests for media derivatives somewhere? Probably not for this one, but we should have one.
> 
> Probably not. Could you invent one test model?

Here is a test model. After simulation, s and s_from_der should be identical. 
Only problem: the test does not pass, and I am unsure whether my model is wrong or the PR.
If I run the identical test for enthalpy, it works just fine.

```Modelica
model sder_test
  replaceable package Medium =
      Modelica.Media.Air.MoistAir;

  Real der_T = 300;
  Real T = 273.15 + der_T * time;
  Real der_p = 2e5;
  Real p = 1e5 + der_p * time;
  Real[Medium.nS] X = Medium.reference_X;
 
  Real s = Medium.s_pTX(p, T, X);
  Real s_der = Medium.s_pTX_der(p, T, X, der_p, der_T, {0,0});
  Real s_from_der;
  
  Real s_error =  s_from_der - s;
  Real s_error_rel =  abs(s_error / s);
initial equation 
  s_from_der = s;
equation 
  der(s_from_der) = s_der;
end sder_test;
```

finback-at, 2019-05-10:
For MoistAir.s_pTX and MoistAir.s_pTX_der, there are other issues #2872 and #2874.
I think It's not better to use for the test.

thorade, 2019-05-10:
But it seems MoistAir is the only place where s0_Tlow and s0_Tlow_der are actually called.
https://github.com/modelica/ModelicaStandardLibrary/search?q=s0_Tlow&unscoped_q=s0_Tlow
All other gases use s0_T (without the low). So, in addition to all other issues, it seems that the names and description strings are misleading.

beutlich, 2019-05-13:
> So, in addition to all other issues, it seems that the names and description strings are misleading.

@thorade Can you please provide a PR for it.

beutlich, 2019-05-14:
> Only problem: the test does not pass, and I am unsure whether my model is wrong or the PR.

@finback-at You are right. If both #2919 and #2932 are merged then @thorade's test model sder_test gives similar results for s and s_from_der.

[image]

beutlich, 2019-07-24:
> 
> 
> > So, in addition to all other issues, it seems that the names and description strings are misleading.
> 
> @thorade Can you please provide a PR for it.

@thorade This is still open, right?

PR #2919 (2019-07-26): Fix algorithm of IdealGases.Common.Functions.s0_Tlow_der
@finback-at @thorade FYI

Fix #2870, fix #1909.
PR #3071 (2019-09-26): maint/3.2.3: Fix algorithm of IdealGases.Common.Functions.s0_Tlow_der
Back-port of #2919 to fix #2870 in maint/3.2.3 branch.
closed 2019-07-26 by beutlich