modelica/ModelicaStandardLibrary #2873: Incorrect parameter in Modelica.Mechanics.MultiBody.Parts.Body?

svorkoetter, 2019-04-02:
`Modelica.Mechanics.MultiBody.Parts` contains the following model (edited for clarity):

    model Body
        ...
        final parameter SI.AngularAcceleration z_a_start[3] = Frames.resolve2(R_start, z_0_start);
        ...
        SI.AngularAcceleration z_a[3](start = Frames.resolve2(R_start, z_0_start), fixed = fill(z_0_fixed, 3));
        ...
    end Body

Shouldn't that second declaration be as follows?:

    SI.AngularAcceleration z_a[3](start=za_start, fixed=fill(z_0_fixed, 3));

Parameter `z_a_start` isn't actually used anywhere at the moment.

tobolar, 2019-04-03:
I'm not absolutely sure since I haven't found any corresponding commit, but I suppose `z_a_start` is leftover from some previous library version in order to not brake backward compatibility (therefore with prefix "final"). So that parameter can be deleted from v4.0.0.

The transformation `Frames.resolve2` shall be correct since vector `z_0_start` is given in world frame whereas `z_a` is in frame_a.

beutlich, 2019-04-04:
git/svn blame is your friend here. When MultiBody was added to MSL 2.1 by ee9a1a29568c2815a0e568719f25482a50d388c7 it already was final: https://github.com/modelica/ModelicaStandardLibrary/blob/ee9a1a29568c2815a0e568719f25482a50d388c7/Mechanics/MultiBody/parts.mo#L625-L628 and both `w_a_start` and `z_a_start` were assigned depending on `initType`: https://github.com/modelica/ModelicaStandardLibrary/blob/ee9a1a29568c2815a0e568719f25482a50d388c7/Mechanics/MultiBody/parts.mo#L827-L840
In 901dd3068dcc071cb2b4a1e1261e4aa0556977fb (for MSL 3.0.1) both `initType` and `w_a_start` have been removed, but `z_a_start` survived - unintentionally, I assume. Since there cannot be any modifier to `z_a_start` we can savely remove it for MSL 4.0.0 (as `w_a_start` was removed). Will prepare a PR.

PR #2875 (2019-04-04): Remove superfluous final parameter Parts.Body.z_a_start
Fix leftover when refactoring MultiBody initialization by 901dd3068dcc071cb2b4a1e1261e4aa0556977fb

close #2873
closed 2019-04-04 by beutlich