modelica/ModelicaStandardLibrary #2892: Is InitPID.DoNotUse_InitialIntegratorState still needed?

beutlich, 2019-04-09:
For MSL 4.0.0, can we remove the PID initialization type [InitPID](https://doc.modelica.org/Modelica%203.2.3/Resources/helpDymola/Modelica_Blocks_Types.html#Modelica.Blocks.Types.InitPID) with the Modelica 2.2 compatible initialization alternative `DoNotUse_InitialIntegratorState`? Or is it still needed?

beutlich, 2019-04-09:
See also #1215. Seems we never can get rid of our legacy.

dietmarw, 2019-04-10:
I'd argue that if there is a broken issue that can not be solved neither in a backward compatible way nor by conversion then we simply need to decide to either live with it forever or fix it in a breaking way. For the latter the MSL 4.0.0 would be the right time. Otherwise it will have to live in the MSL for another 10 years at least which seems not appropriate.

MartinOtter, 2019-04-10:
Fine for me to remove option DoNotUse_InitialIntegratorState for MSL 4.0.0, even if it is not possible to provide a conversion script.

PR #2898 (2019-06-30): Remove legacy initialization of Modelica.Blocks.Continuous.{PID, LimPID}
@HansOlsson This is what I tried to remove the now invalid modifier `Init.DoNotUse_InitialIntegratorState`. But no matter what I did, it did not work out, since enumeration simplification did not run when converting `ModelicaTestConversion4.Blocks.Issue2892`.

```
convertModifiers({"Modelica.Blocks.Continuous.LimPID",
                  "Modelica.Blocks.Continuous.PID"},
                  {"initType"}, {"initType=if Integer(%initType%) < 5 then %initType% else Modelica.Blocks.Types.Init.NoInit"}, true);
```

Do you have an [... cut]
closed 2019-06-30 by beutlich