modelica/ModelicaStandardLibrary #2899: Remove components having "Heating" in their name

christiankral, 2019-04-13:
Having components with `Heating` in their name are usually a leftover from the time when components were designed without heat port. Afterwards, the `Heating` components were add to the MSL. Later, step by step, the heat ports were added in a backwards compatible way. Therefore, all these `Heating` components shall be removed from the MSL:

### Components

- `Modelica.Electrical.Analog.Basic.HeatingResistor`
- `Modelica.Electrical.Analog.Semiconductors.HeatingDiode`
- `Modelica.Electrical.Analog.Semiconductors.HeatingNMOS`
- `Modelica.Electrical.Analog.Semiconductors.HeatingNPN`
- `Modelica.Electrical.Analog.Semiconductors.HeatingPMOS`
- `Modelica.Electrical.Analog.Semiconductors.HeatingPNP`

### Examples

In the following examples the `Heating` components shall be replaced and the term `Heating` shall be replaced in the example name. 

- `Modelica.Electrical.Analog.Examples.HeatingMOSInverter`
- `Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate` shall be rename to `NPN_NORGate` as proposed in #2882
- `Modelica.Electrical.Analog.Examples.HeatingResistor`
- `Modelica.Electrical.Analog.Examples.HeatingRectifier`
- `Modelica.Electrical.Analog.Examples.HeatingResistor`

dietmarw, 2019-04-13:
Just be aware that some of those components don't just have heatports but also have different (heat dependent) characteristics. So the original non-Heating components might need to get extended, as in some features might need to get merged back into the normal components.

christiankral, 2019-06-21:
Having a closer look:

christiankral, 2019-06-21:
## Diodes in `Modelica.Electrical.Analog.Semiconductors`

`Diode` | `Diode2` | `HeatingDiode` | Description
--------- | ----------- | ------------------- | --------------
-- | -- | `EG` | Activation energy
`Ids` | `Ids` | `Ids` | Saturation current
`Maxexp` | `Maxexp` | `Maxexp` | Max. exponent for linear continuation
-- | `N` | `N` | Emmision coefficient
`R` | -- | `R`| Parallel ohmic resistance
 -- | `Rs` | -- | [Series] Ohmic resistance
`T` | `T`| `T`| Fixed device temperature if useHeatPort = false
-- | -- | `TNOM` | Parameter measurement temperature
`useHeatPort` | `useHeatPort` | `useHeatPort` | =true, if HeatPort is enabled
-- | `Vf`| -- | Forward voltage
`Vt`| `Vt` | -- | Temperature voltage
-- | -- | `XTI` | Temperature exponent of saturation current

So it seems all the three models have a different operation behavior and cannot (easily) be converted into one model. @christophclauss @kristinmajetta My proposal is here to rename ``HeatingDiode` e.g. into `Diode3` or a more meaningful name.

christiankral, 2019-06-21:
## NMOS and PMOS in `Modelica.Electrical.Analog.Semiconductors`

A comparison of the PMOS and NMOS models shows many equally named parameters. The following tables shows only the parameters which are not present in each of the four models:

 `NMOS` and `PMOS` | `HeatingNMOS` and `HeatingPMOS` | Description
--------------------------- | ----------------------------------------------  | -------------
-- | `Tnom` |  Parameter measurement temperature
-- | `kvt` | Fitting parameter for Vt
-- | `kk2` |  Fitting parameter for K2

So to me this looks like that the "Heating" components are more elaborate than the others. So possibly the `NMOS` and `PMOS` can be substituted by `HeatingNMOS` and `HeatingPMOS`. As I am no expert in electronics, I am asking @christophclauss and @kristinmajetta to comment here.

christiankral, 2019-06-21:
## NPN and PNP in `Modelica.Electrical.Analog.Semiconductors`

I guess the situation is similar to that of the diodes. The two model types `ǸPN` (and `PNP`) differ from `HeatingNPN` (and `HeatingPNP`) in such way that some parameters do not overlap. So possibly these models have be kept separately due their different operational behavior. I suspect that the "Heating" of this model shall be replaced by something more meaningful or simply adding a number to the name, like `NPN2`. @christophclauss and @kristinmajetta can possibly make a comment here.

christiankral, 2019-06-21:
## Resistors in `Modelica.Electrical.Analog.Basic`

The two resistor models `Resistor` and `HeatingResistor` have the very same operational behavior but different parameter names of the reference resistance at the reference temperature.

 `Resistance` | `HeatingResistance` | Description
--------------------------- | ----------------------------------------------  | -------------
`R` | -- | Resistance at temperature T_ref 
-- | `R_ref` | Resistance at temperature T_ref

I will create a PR to migrate `HeatingResistor` into `Resistor` if I can create a working conversion script.

- Step 1: Replace `HeatingResistor` by `Resistor` in all user models:

```modelica
convertClass("Modelica.Electrical.Analog.Basic.HeatingResistor",
              "Modelica.Electrical.Analog.Basic.Resistor");
```

- Step 1: The remaining parameters `R_ref` needs the to be converterd into `R` in `Resistor`:

```modelica
convertElement("Modelica.Electrical.Analog.Basic.Resistor",
               {"R_ref"},
               {"R"});
```

I suppose this shall be possible. If I apply the two conversions in the given order I can avoid messing with the order of conversion statements in the script: 

1. `convertClass`
1. `convertElement`
1. `convertModifiers`

@beutlich Do you agree?

beutlich, 2019-06-21:
Yes, sounds reasonable.

christophclauss, 2019-06-21:
Converting the HeatingResistor as you suggested should work. It is o.k. to me.
For checking the Diode, MOS as well as Bipolar models I need some time. Possibly simplifications can be done. First I will look at the equations.

christophclauss, 2019-06-21:
Sorry for closing!

beutlich, 2019-06-21:
> For checking the Diode, MOS as well as Bipolar models I need some time.

Can you estimate how long it will take as I do not want to see the current momentum stopping again? Thanks.

christophclauss, 2019-06-22:
I also do not like stopping the current initiatives. I will do it within the next 14 days, starting with the diodes.

christophclauss, 2019-06-27:
To diodes:
HeatingDiode and Diode are very similar, we can combine both models to the model calls Diode. Diode2 should remain as it is.
I suggest the following:
Within the MSL
1) The Diode model is deleted.
2) The HeatingDiode model gets a further parameter: 
Boolean DependsOnTemperatureOfHeatPort, which is false by default. If this parameter is false, the temperature T of the heat port is not used, the Heating Diode behaves like the deleted diode whithout heating. If it is true, the temperature T of the heat port is used.
3) Some small changes in the code are necessary.
4) The HeatingDiode is renamed into Diode.

Within the applications
1) The usage if Diode is not changed.
2) The usage of HeatingDiode is changed into Diode. The parameter DependsOnTemperatureOfHeatPort=true is added.  (I'm not sure, if this is possible automaticly)

If you (Thomas, Christian) agree, I develop  and check the new diode model.

beutlich, 2019-06-28:
I also checked the current behaviour of both Diode and HeatingDiode.

* A Diode can be **exactly** modelled by a HeatingDiode, **only** if useHeatPort=false (and N=Diode.Vt*q/k/T, EG=0, XTI=0).
* A Diode cannot be exactly modelled by a HeatingDiode if useHeatPort=true, since the diode current (HeatingDiode.id) depends on the temperature, whereas it is does not for the Diode.
* Therefore, as you proposed, it seems right to introduce a new Boolean parameter (DependsOnTemperatureOfHeatPort or better: useTemperatureDependency) to switch the behaviour of the temperature dependency.

Thus, basically, what you propose is to rename HeatingDiode to Diode, i.e.,

```
convertClass("Modelica.Electrical.Analog.Semiconductors.HeatingDiode",
             "Modelica.Electrical.Analog.Semiconductors.Diode");
```

which is analog to the renaming of HeatingResisitor to Resistor by #2983. If we set the default value of useTemperatureDependency to false, we simply can apply the following conversion for the HeatingDiode

```
convertModifiers({"Modelica.Electrical.Analog.Semiconductors.HeatingDiode"},
                  fill("",0), {"useTemperatureDependency=true"});
```

provided, the new Diode model has a parameter Vt (to be used if useTemperatureDependency=false) and uses the parameters N, EG, TNOM, XTI only if useTemperatureDependency=true.

Summarizing: Yes, that should be possible and I already tested the above conversion.

beutlich, 2019-06-28:
@christophclauss Can you please analyze the NMOS, PMOS and NPN, PNP conversion next. Thanks.

beutlich, 2019-06-28:
@christophclauss To save you some work I created a draft PR #3010 with the actual equationsleft to do.

christophclauss, 2019-07-02:
To PMOS
The PMOS model can be easily merged with the HeatingPMOS model if the following steps are done:

Within MSL:
1. The PMOS model is deleted.
2. The HeatingPMOS model gets further parameter useTemperatureDependency like in the Diode model, which is false by default. If this parameter is false, the temperature of the heat port is not used, the HeatingPMOS behaves like the former PMOS model. If it is true, the temperature of the HeatPort is used for calculating the current.
3. In the HeatingPMOS model the equations for calculating beta_t, vt_t, and k2_t are modified to:
`beta_t  = if useTemperatureDependeny then Beta*pow((T_heatPort/Tnom), -1.5 else Beta;`
`vt_t =  if useTemperatureDependeny then Vt*(1 + (T_heatPort - Tnom)*kvt else Vt`
`k2_t =  if useTemperatureDependeny then K2*(1 + (T_heatPort - Tnom)*kk2 else K2`
4. The HeatingPMOS model is renamed to PMOS.

Within the applications
    The usage if PMOS is not changed.
    The usage of HeatingPMOS is changed into PMOS. The parameter useTemperatureDependency=true is added. 

The NMOS transistor model should be changed analogously.

christophclauss, 2019-07-15:
To NPN
Also the NPN model can be merged with the HeatingNPN model by the following steps:

Within MSL:
==========

    The NPN model is deleted.
    The HeatingNPN model gets the further parameter useTemperatureDependency like in the Diode model, which is false by default. If this parameter is false, the temperature of the heat port is not used, the HeatingNPN behaves like the former NPN model. If it is true, the temperature of the HeatPort is used for calculating the currents.
    In the HeatingNPN model the equations for calculating is_t, bt_t, bf_t, and vt_t are modified to:
    is_t = if useTemperatureDependeny then Is*pow((T_heatPort - Tnom), XTI)*htempexp else Is;
    br_t = if useTemperatureDependeny then Br*pow((T_heatPort - Tnom), XTB) else Br;
    bf_t = if useTemperatureDependeny then Bf*pow((T_heatPort - Tnom), XTB) else Bf;
    vt_t = if useTemperatureDependeny then (k/q)*T_heatPort else Vt;

 Two additional variables are introduced: Real nr_t, nf_t;
 Two equations are added:
   nr_t = if useTemperatureDependeny then NR else 1;
   nf_t = if useTemperatureDependeny then NF else 1;
 Within the equations, which calculate ibc, ibe, cbc, and cbe:
   NR is replaced by nr_t
   NF is replaced by nf_t

    The HeatingNPN model is renamed to NPN.

Within the applications:
The usage if NPN is not changed.
The usage of HeatingNPN is changed into NPN. The parameter useTemperatureDependency=true is added.

The PNP transistor model will be changed analogously.

PR #3178 (2019-10-29): Merge HeatingNPN/HeatingPNP with NPN/PNP
This is the last missing conversion to finally close #2899.
closed 2019-10-29 by beutlich