modelica/ModelicaStandardLibrary #2901: Provide higher order derivatives for CombiTables

hyumo, 2019-04-18:
It would be appreciated if higher order derivatives are provided in the standard tables.

thorade, 2019-04-23:
To my understanding, MSL tables are C1 smooth, so you can only get first derivatives really. What order would you need?

hyumo, 2019-04-23:
In my case, I need at least C2, and fundamentally it should be possible to provide up till C3 with the current spline type, it is just not implemented in standard tables.
I am not sure if there's any historical reason why C2 and C3 derivative functions are not provided (they are available in Dymola dymtable before ModelicaStandardTable was introduced), if there are, please point me to the related issue, thanks.

Maplesoft-fmigroup, 2019-04-24:
Seconded!
For our MapleSim product we convert to use our own implementation
of interpolation, for which the differential order is an argument
to the interpolation function.
Any non-negative integer is valid for the diff. order, where high
enough values (i.e. larger than the degree of the underlying
polynomial used for the interpolant) simply give a function that
always returns zero.

hyumo, 2019-04-24:
Same here, I added the higher order derivatives myself in ModelicaStandardTables.c and recompiled it as a new static lib, which is really just a hack... and if I could do this, I believe ModelicaStandardTables could do this in a much better way.

Maplesoft-fmigroup, 2019-04-24:
We had our own implementation of interpolation that included the 
differential order and a number of options for extrapolation that were 
not present in the original MSL implementation (I think they have added 
these extrapolation options since).

Computing the derivative of an interpolant is a fundamentally simpler 
operation than the interpolant itself, and far less error prone than 
attempting numerical differentiation.

- Allan

beutlich, 2019-05-03:
> 
> 
> Same here, I added the higher order derivatives myself in ModelicaStandardTables.c and recompiled it as a new static lib, which is really just a hack... and if I could do this, I believe ModelicaStandardTables could do this in a much better way.

Can you share your diff please. Thanks.

Currently, only C1 is guaranteed (due to the cubic Hermite spline interpolation). See my example where the second derivative no longer is continuous (either due to the extrapolation or to the data itself). In your implementation you should be able to replace the approximated second derivative by the symbolic second derivative.

```mo
model T2901
  Modelica.Blocks.Sources.CombiTimeTable combiTimeTable1(
    table=[0.25, 20; 0.5, 0; 0.75, 10],
    smoothness=Modelica.Blocks.Types.Smoothness.ContinuousDerivative) annotation(Placement(transformation(extent={{-130,65},{-110,85}})));
  Modelica.Blocks.Continuous.Der der1 annotation(Placement(transformation(extent={{-100,65},{-80,85}})));
  Modelica.Blocks.Continuous.Derivative derivative1(
    T=0.001,
    initType=Modelica.Blocks.Types.Init.InitialOutput) annotation(Placement(transformation(extent={{-70,65},{-50,85}})));
  Modelica.Blocks.Sources.CombiTimeTable combiTimeTable2(
    table=[0.0, 20; 0.3, 0; 0.6, 0; 1.0, 10],
    smoothness=Modelica.Blocks.Types.Smoothness.MonotoneContinuousDerivative2) annotation(Placement(transformation(extent={{-130,35},{-110,55}})));
  Modelica.Blocks.Continuous.Der der2 annotation(Placement(transformation(extent={{-100,35},{-80,55}})));
  Modelica.Blocks.Continuous.Derivative derivative2(
    T=0.001,
    initType=Modelica.Blocks.Types.Init.InitialOutput) annotation(Placement(transformation(extent={{-70,35},{-50,55}})));
  equation
    connect(combiTimeTable1.y[1],der1.u) annotation(Line(points={{-109,75},{-104,75},{-107,75},{-102,75}}, color={0,0,127}));
    connect(der1.y,derivative1.u) annotation(Line(points={{-79,75},{-74,75},{-77,75},{-72,75}}, color={0,0,127}));
    connect(combiTimeTable2.y[1],der2.u) annotation(Line(points={{-109,45},{-104,45},{-107,45},{-102,45}}, color={0,0,127}));
    connect(der2.y,derivative2.u) annotation(Line(points={{-79,45},{-74,45},{-77,45},{-72,45}}, color={0,0,127}));
end T2901;

```

Nevertheless, the second derivatives can be provided indeed (w/o being C2).

Of course, we could add cubic spline interpolation being C2 (but these would tend to greater overshoots and/or no longer be shape preserving). This is a different story then and rather related to https://github.com/modelica/ModelicaStandardLibrary/issues/1153#issuecomment-480260350,

beutlich, 2019-08-09:
> where high enough values (i.e. larger than the degree of the underlying
> polynomial used for the interpolant) simply give a function that always returns zero.

As discussed in #1147, it is not legal (or desired?) in MSL to provide constant zero derivative functions for (potentially) discontinuous functions (as is the case with cubic Hermite C1 splines).

#3083 introduces the second derivative functions consistently. Note, that no time events are introduced for the case of a discontinous second derivative of a CombiTimeTable.

PR #3083 (2019-10-20): Add second derivatives to table blocks
Close #2901.
closed 2019-10-20 by beutlich