modelica/ModelicaStandardLibrary #2903: Off-by-one Error Seg Fault in ModelicaStandardTables.c

svorkoetter, 2019-04-24:
Line 26 of `Modelica/Blocks/Tables.mo` has:

    parameter Integer columns[:]=2:size(table, 2);

Line 44 initializes an external object, as defined on lines 179-213 of `Modelica/Blocks/Types.mo`. The `columns` parameter, as well as `size(columns,1)`, is passed to the constructor.

The code on line 1868 of `Modelica/Resources/C-Sources/ModelicaStandardTables.c` allocates space for the columns based on the size. Thus for an N column table, it allocates a vector of N-1 integers (i.e., `size(table,2)-2+1`).

The code in `ModelicaStandardTables_CombiTable1D_getValue`, on line 1976 of the same file, has:

    const size_t col = (size_t)tableID->cols[iCol - 1] - 1;

I believe that `iCol - 1` should be `iCol - 2`. Since the lowest value for iCol on the Modelica side is 2, and C arrays are zero-based. The way the code is now, accessing the last column indexes past the end of the `cols` array, giving a segmentation fault.

Maplesoft-fmigroup, 2019-04-25:
Same issue appears in`ModelicaStandardTables_CombiTable1D_getDerValue`, `ModelicaStandardTables_CombiTimeTable_getValue`, and `ModelicaStandardTables_CombiTimeTable_getDerValue`.

svorkoetter, 2019-04-30:
It seems to have been an error on our part in understanding the code.

closed 2019-04-30 by svorkoetter