modelica/ModelicaStandardLibrary #2908: Staropint in Electrical.Machines.Unilities.TerminalBox

max-privato, 2019-04-30:
If I hover with mouse on starpoint connector in TerminalBox I see as tooltip, 

_in Dymola:_    [...] startpoint if (terminalConnection "D") 

_in OpenModelica:_ starpoint

in both cases I miss the real information which, IMO should be "starpoint, used only if connection is "Y" (default)".

This occurs because starpoint has not a descriptive text. What I propose is to substitute, in TerminalBox:
```
 Modelica.Electrical.Analog.Interfaces.NegativePin starpoint if (
    terminalConnection <> "D")

```
with 
```
 Modelica.Electrical.Analog.Interfaces.NegativePin starpoint if (
    terminalConnection <> "D") "starpoint, used only if connection is 'Y' (default)"

```
this will improve thins, even if in Dymola the tooltip still remains partly strange, because the two characters '<>'  between terminalConnection and "D" are not displayed .

christiankral, 2019-06-16:
For some reason it's already included in the MSL master of `Modelica.Electrical.Machines.Utilities.TerminalBox` and `Modelica.Electrical.Machines.Utilities.MultiTerminalBox`:

```
Modelica.Electrical.MultiPhase.Interfaces.NegativePlug starpoint(final m=
      mSystems) if (terminalConnection <> "D") "Star point"
```

I think the text is sufficient and does not require the proposed and redundant information that the star point is only there, when the terminal box uses a star connection.

closed 2019-06-16 by christiankral

beutlich, 2019-06-16:
Looks like a tool issue that `<>` is considered as empty HTML tag wehen rendering the rool tip:

[image]

max-privato, 2019-06-17:
> For some reason it's already included in the MSL master of `Modelica.Electrical.Machines.Utilities.TerminalBox` and `Modelica.Electrical.Machines.Utilities.MultiTerminalBox`:
>
> ``` [...]

Ah, ok. The thing is that OM presently shows conditional connectors. It shouldn't and there are plans to correct this, so I agree with leaving the short descriptive text.