modelica/ModelicaStandardLibrary #2911: Issue with SM_PermanentMagnet?

max-privato, 2019-05-02:
I'm not totally sure, but I think there is an issue with the management of Lssigma in Modelica.Electrical.Machines.SynchronousInductionMachines.SM_PermanentMagnet model.

I expect that at equal terminal currents, the terminal voltage is influenced by the Lssigma value.
The built-in start value is 0.1 ohm (318 uH), as large as one third of Lmd=Lmq. The voltage drop across it should be significant. 
If, however, the Lssigma value is changed from the built-in start value (318 uH) into a much lower value (say 1e-7H), the terminal voltage shows no change. Interesting enough, the voltage across smpm.lssigma (lsssigma.v_) shows substantial changes.
I really cannot explain this. Am I wrong or there is something wrong with the model?

This behaviour can be replicated using the model accessible here:
https://unipiit-my.sharepoint.com/:u:/g/personal/a008042_unipi_it/EfxIERNQ8olPqTmcwxk2CqkBASj_d3vtehFWpUJWytEx5w?e=gOzXaK

max-privato, 2019-05-03:
Need to think more about it...

closed 2019-05-03 by max-privato

christiankral, 2019-05-03:
If you compare the terminal voltages of the machine with different values of `Lssigma`, you are comparing RMS or peak values, respectively. But there exists a phase shift of the voltage drop across `Lssigma` compared to the terminal voltage depending on the operating conditions. Whenever you plot the voltage drops of the phasor `smpm.lssigma.v_[..]` you can observe the expected voltage drop across the stray inductors.

Conseuquently, changing `IdRef` may also more or less influence the RMS value of the terminal voltage.

max-privato, 2019-05-03:
Thank you. 
I closed the ticket just to probe the effects of Id more .
In fact, the effects on terminal voltage of presence or absence of Lsigma might be evanesent if voltages created by PMs and by stator currents are in quadrature (what happens exactly if I have just Iq). Using my model with some Id (e.g. 50 A), shoes reasonable voltage difference upon changing Lssigma by an order of magnitude.
Thanks again for you comment.