modelica/ModelicaStandardLibrary #2921: Missing derivative annotation pointing to MixtureGasNasa.h_TX_der?

henrikt-ma, 2019-05-08:
I looks to me as if `Modelica.Media.IdealGases.Common.MixtureGasNasa.h_TX_der` is unused, but that it was meant to be used as the derivative of `Modelica.Media.IdealGases.Common.MixtureGasNasa.h_TX`.

I've tried adding the `derivative(zeroDerivative = h_off) = h_TX_der` annotation, and that seems to be working fine with the regression test for `Modelica.Media.Examples.MixtureGases`.

beutlich, 2019-05-08:
See #1909 for the motivation of smoothOrder (that is set for h_TX_der).

henrikt-ma, 2019-05-08:
OK, that's a good reason to not use `h_TX_der`.  Then I'm just not able to see the point of still having it in the library.

beutlich, 2019-05-09:
Good question, sincer there also is s_TX but no s_TX_der.

beutlich, 2019-07-24:
@thorade Can you please have a look here please? Thanks.

thorade, 2019-07-25:
These derivatives might have been called explicitly somewhere in the past, maybe mixtures, similar to the situation with `h_Tlow_der`:
https://github.com/modelica/ModelicaStandardLibrary/pull/2919#issuecomment-490215734

But today, searching for `h_TX_der` in the library gives zero results, they are just not used anywhere, and I assume it is safe to delete them.

qlambert-pro, 2021-06-02:
A similar issue exists with `Modelica.Media.Water.IF97_Utilities.rho_pT` where it has no relevant annotation but `Modelica.Media.Water.IF97_Utilities.rho_pT_der` exists.

thorade, 2021-06-02:
My assumption here is: Dymola and other tools are able to automatically calculate the time derivative, so there is no need to add the annotation. Instead, the derivative functions were called explicitly, maybe from a volume dynamics model, or from a mixture model. All we can check now, is whether the derivative functions are called from anywhere in the library, but we just cannot know whether any user is using them.