modelica/ModelicaStandardLibrary #2923: Oddly located name and connector of Modelica.Electrical.Analog.Sources.SignalVoltage icon

christoff-buerger, 2019-05-09:
Compared to MSL 3.2.2, the Icon of `Modelica.Electrical.Analog.Sources.SignalVoltage` changed in MSL 3.2.3. There are two changes that seem unnecessary and result in odd looking diagrams:
 1. The `v` input connector is hanging in the air, not close and above the rest of the icon. In MSL 3.2.2 it was next to the ellipse/rest of the icon.
 2. The name is moved in-between the `v` input connector and the ellipse, sort of above the actual icon. In MSL 3.2.2 it was just below the icon.

I am wondering, are these changes intentional? It looks odd that the name is squeezed in-between the "actual" icon and an input connection. In diagrams the name locations are now flipped and the `v` input connections seem to hang in the air.

The same changes are done for `SignalCurrent`. I can understand, that we may like to have all names above icons, but at the price to break off input connections that are then hanging in the air? This looks really ugly in diagrams -- actually it looks like a pending connection error either due to a connection not properly drawn but correctly modeled (i.e., in fact connected but not shown as such) or indeed with a loose end (i.e., connected to soemething that does not exist and just drawn).

beutlich, 2019-05-11:
Since a picture says more than words:

[image]

The changes are due to the clarified icon conventions and performed in #2215.

christiankral, 2019-06-20:
There are always other options to design icons. One big achievement of in the icon design guidelines was to always have the connector on the same place and to have the component name and possible parameters at the same place. This enables a unique look and unique design of models. Since we agreed on all these design guidelines I would definitely not like to change these guidelines, as this would either cause all the components NOT look uniquely again or to re-design the icons of the entire MSL which wll then cause some "this is not the ideal appearance" cases at some other places. 

The original intention was to create a simple icon, which follows simple design rules which are stated in MSL User's Guide. Alternatives:

[image]

- `signalVoltage1`: Currently used in MSL 3.2.3
- `signalVoltage2`: Design is in line with design guidelines in `Modelica.UsersGuide.Conventions.Icons` and looks actually good, but moving the connector to the bottom will visually mess up all user models; in case the icon design were flipped vertically, it were no longer in line with icon design guidelines
- `signalVoltage3`: Same as `signalVoltage2`
- `signalVoltage4`: Design is in line with icon guidelines but the line interferes with text and makes sometimes hard to read it, as in 
[image]

- `signalVoltage5`: Dotted line similar to `signalVoltage4`, but not better to read
- `signalVoltage6`: Gray line similar to `signalVoltage4`, but not better to read

My conclusion is: Number 2 and 3 either mess up all user models or are not in line with the design guidelines and the number 4 to 6 are very bad to read. My proposals are: 
1. Leave it as it is (my preference). 
2. Make a proposal being in line with the icon design guidelines where lines do do interfere with text: if we agree on the design then you can create a PR for `Electrical.{Analog|MultiPhase|QuasiStatic.{SinglePhase|MultiPhase}}` and `Magnetic.{FluxTubes|QuasiStatic.FluxTubes}`

christoff-buerger, 2019-06-24:
Maybe I should clarify that I did not open this issue just because I dislike the new icon. I got issues because the new icon can result in misguiding rendering of existing model-diagrams if they are not manually adjusted (thus, it can "visually mess up existing user models" in a subtle way).

The scenario I had was the following MSL 3.2.2 diagram (excerpt):

[image]

Loading that model in MSL 3.2.3 shows:

[image]

This is a misleading rendering, looking as if the output of `voltageController` is directly connected to `excitationVoltage`.

It is a good _intent_ to have all component names at the very same place according to some guideline; but, I think, if this results in disconnected connections just to satisfy the guideline, this is not a good _idea_. I mean, if we have three connections on three different sides, most user applications will just vertically and horizontally flip the component in its application contexts to avoid cumbersome connections. Such flipping will (should/must) take care of the component name; there is no reasonable way to enforce that the name is always above the component. It is much more important that the connections are fitting. In Dymola, we can adjust the component according to its usage context by flipping as follows:

[image]

This is what will happen most times: to arrange the in- and outputs to fit the application context.

For that reason I think that `signalVoltage2` is the way to go. Not in the spirit of
 > signalVoltage2: Design is in line with design guidelines in Modelica.UsersGuide.Conventions.Icons and looks actually good, but moving the connector to the bottom will visually mess up all user models; in case the icon design were flipped vertically, it were no longer in line with icon design guidelines

but rather
 - Design is in line with design guidelines in Modelica.UsersGuide.Conventions.Icons and looks actually good. The component name location is on the one side without connector, because the component will anyway be flipped in application contexts to avoid complicated connection-lines that have to circumvent the component and cross-cut each other.

christiankral, 2019-06-24:
First, I agree that `signalVoltage2` is the best choice. However, for newly designed models it is not issue to flip and rotate the model. But existing models may be messed up, e.g. `Modelica.Electrical.Analog.Examples.GenerationOfFMUs`:

[image]

Second, for the example you presented:

[image]

No surprise: whenever icons change, they will look different. And yes, a changed look of an icon affects the appearance of every user model. So to me your example is a typical case, where the voltage source needs some additional horizontal spacing to the right to make it look good again. So with `signalVoltage4` to `signalVoltage4` it would also look the same, even though the connector were attached to the circle by a line.

**Summary:** Either leave everything as it is (my preference) or switch to `signalVoltage2` (with the risk of messing up user models): however, I think we some kind of majority here if we go for the `signalVoltage2` implementation.

christoff-buerger, 2019-11-04:
> Summary: Either leave everything as it is (my preference) or switch to signalVoltage2 (with the risk of messing up user models): however, I think we some kind of majority here if we go for the signalVoltage2 implementation.

I would vote for `signalVoltage2 `; but as you say, it likely requires adjustment of existing models, where its applications have to be flipped once (for example vertically in your example).

If I am the only one annoyed by the new unconnected icon design, then I guess, it's as you say: leave everything as it is. Considering the little response so far, I say just close the ticket.

It would be nice in the future though, if relentless following the design guide lines would consider practice also and avoid icons that fall apart with dangling connectors. As I stated above, one would anyway flip the component until its connections fit the application context; a rule like _"component names must always be above the component"_ makes in the context of icons with only one connector-free side little sense since respective components will be flipped all the time anyway. It is the  connector-free side the component name should be; if one wants the component name above, fine -- but put the 3 connectors on the other sides.

closed 2020-01-24 by dietmarw