modelica/ModelicaStandardLibrary #2931: Proposal: Styling Guidelines of Modelica Code

christiankral, 2019-05-14:
# Proposal: Styling Guidelines of Modelica Code

Some programming languages have styling guidelines which recommend how code shall be written and formatted. These styling guidelines help to make the code appear in a uniform way.

For Modelica such styling guidelines do not yet exist. Introducing such guidelines would have some advantages on my opinion. This includes the following aspects:

## Code Format

Modelica tools might support some auto-format of code depending on the styling guidelines (if enabled in the tool). If each tool obeys the same styling guidelines, it could be avoided that using different tools could cause changed line breaks, spaces, etc.

Such styling guidelines (or rules) might include (and may be extended):

- Blanks and other white spaces
- Indentations
- Line breaks  

Possible example of good style

```
a + b = c;
z = sqrt(x^2 + y^2);
for k in 1:n loop
    x[1, k] = 3 * k;
end for;
```

Possible example of bad style

```
a+b=c;
z=sqrt(x^2+y^2);
for k in 1 : n loop
  x[1,k]=3*k;
end for;
```

Such code formatting guidelines and / or rules shall then also be included and implemented in the Modelica Language Specification document.  

## Variable Naming Conventions

Long time ago an attempt was made to somehow provide a
[naming convention](https://doc.modelica.org/Modelica%203.2.3/Resources/helpOM/Modelica.UsersGuide.Conventions.ModelicaCode.Naming.html)
of variables. At the time it was created, it already was a compromise as it was not possible to introduce any non-backwards compatible changes at the time. Therefore we specified variables like `HFlow` and `Qflow` which are obviously spelled according to different rules, as the `H` in `HFlow` is uppercase and the `f` in `Hflow` is lowercase.   

The conversion scripts of MSL 4.0.0 provides the opportunity to us to streamline the variable naming. Proposal [#2357 @ Modelica Specification](https://github.com/modelica/ModelicaSpecification/issues/2357) might additionally support an improved naming convention, however. A new possible variable naming convention might follow some simple rules and [#2357 @ Modelica Specification](https://github.com/modelica/ModelicaSpecification/issues/2357) might be adapted such way that simple naming conventions might be established.

Possible rules (to be discussed)

- Derive variable names form physical symbols as they are typed in scientific publications:
- Use `_` to separate indices from variables
- Use `´` to indicate time derivatives
- Use ` ⃗` to indicate a vector
- Use a special character, e.g. `@`, to indicate an operating point
- Provide default abbreviations of operating points, e.g. `start`, `end`, `nominal` (instead of `nom`), `max` (instead of `maximum`)

Possible good examples of variable names (to be discussed)
- `α_20` instead of `alpha20`: linear temperature coefficient
- `∆p` instead of `dp`: Pressure drop
- `η` instead of `eta`: Efficiency
- `R_s` instead of `Rs`: Stator resistance of induction machines
- `R_r´` instead of `Rr`: Rotor resistance transformed to stator side
- `V_a@nominal` instead of `VaNominal`: Nominal armature voltage of DC machine
- `ω` instead of `w`: angular velocity
- `t@start` or `t_start` instead of `startTime`: start time
- `∆t` instead of `duration`: time duration
- `τ` instead of `tau`: torque
- `τ@max` or `τ_max` instead of `tauMax`: maximum torque

## Implementation Aspects

Whenever we implement (new) guidelines it shall be done in such a way, that existing code does not get **invalid** automatically. Instead the guidelines shall be **strong** recommendation, not a must.

tobolar, 2019-05-14:
Guidelines: I remember we doveloped some rules within the [EuroSysLib](http://www.eurosyslib.com/) project years ago. But it definitely didn't contain code indention or similar things.

thorade, 2019-05-14:
The interesting part is probably rules that can be enforced by a tool.
Python users for example have PEP8 with rules, and then tools like autopep8. 
https://www.python.org/dev/peps/pep-0008/
https://github.com/hhatto/autopep8

So, I am interested in whitespace before/after operators, line breaks for arguments of function calls, and similar. Also, tools should NEVER modify my code, unless I tell it to do so.