modelica/ModelicaStandardLibrary #2934: Why PowerLoss in LossyGearDemos

HansOlsson, 2019-05-14:
Modelica.Mechanics.Rotational.Examples.LossyGearDemo{1,2} contain  

```
SI.Power PowerLoss=gear.flange_a.tau*der(gear.flange_a.phi) + gear.flange_b.tau
      *der(gear.flange_b.phi) "Power lost in the gear";
```
This is already computed as `gear.lossPower` so it seems unnecessary to duplicate the computation, and even not clear if it should be at the top-level, i.e. could remove the top-level variable.

In addition for Modelica.Mechanics.Rotational.Examples.LossyGearDemo2 it is misleading as the total PowerLoss should add the losses from both gear and bearing.

tobolar, 2019-05-15:
I don't know the backgroud of implementing that variable. I can imagine this is some leftover from times before having `gear.lossPower`. So to me it is fine to delete it. The documentation must be changed accordingly.

Btw: The decription of `PowerLoss` in `LossyGearDemo2` says "Power lost in the gear". But the interpretation of "gear" is ambivalent.

@MartinOtter Any comment?

MartinOtter, 2019-07-19:
I have no strong opinion. Having PowerLoss at the top level has the slight advantage to have it on top level in the variable list (instead of hiding it in a large number of variables). The goal was to show the effect of the efficiency factor in component "gear" (so not the losses of the complete gear box that includes the bearingFriction). The implementation shows that the loss power computed internally in component gear is identical to the difference of the power flowing to flange_a and flange_b. This could be pointed out more clearly in the documentation.

tobolar, 2019-09-03:
@MartinOtter
> The implementation shows that the loss power computed internally in component gear is identical to the difference of the power flowing to flange_a and flange_b. 

Such check would probably be better for some example in "ModelicaTest" library. Since the variable PowerLoss brings no additional benefit, I would prefer to delete it. This is also consistent to `LossyGearDemo3` where no such variable exists.

beutlich, 2019-09-03:
> I would prefer to delete it.

Please prepare a PR for it.

beutlich, 2019-10-15:
> 
> 
> > I would prefer to delete it.
> 
> Please prepare a PR for it.

OK, #3153 resolves it as discussed.

MartinOtter, 2019-10-16:
I have re-inspected LossyGearDemo1/2 and changed my mind:

Internally, variable "tau_loss" is computed in a non-trivial way and from this variable the loss power is computed (tau_loss*w_a). It is easy to have an error in the computation of "tau_loss" and then also the loss power and the whole component is wrong.

I recall that I introduced the loss power computation on the top level to check that it is identical to the sum of the power flowing through the connectors. This is an import check and I would not remove it. Instead the difference of the loss powers should be computed and in the documentation stated that this should be close to zero if the component is correctly implemented.

In LossyGearDemo2 it is clearly stated that the top level variable computes the loss power of "gear" so the component with instance name "gear". It does not make sense at all to compute the whole loss power, because the loss power of "gear" is used to check whether the LossyGear is correctly implemented.

Examples are used to demonstrate how the component is used, but they can also be used at the same time to check if a component is correctely implemented. Therefore, it is fine to keep this variable, provided the documentation points out the reason for it.

beutlich, 2019-10-18:
Reassigning to @MartinOtter to improve the documentation of these examples as proposed.

MartinOtter, 2019-10-18:
> If we just document that the goal is to check that the externally computed loss-power matches the internal one the obvious question would be: why is the internal formula so complicated that we have to check it, especially if the external one looks so simple?

tau_loss is a central variable from which other variables are computed. Only if all variables are computed, the simple loss-power formula can be used.

HansOlsson, 2019-10-18:
> tau_loss is a central variable from which other variables are computed. Only if all variables are computed, the simple loss-power formula can be used.

Both of the given formulas are fairly simple so I don't know which you mean "simple loss-power formula"; and we are as far as I know computing all variables.

However, as indicated above the `PowerLoss` formula would lack a term if `support` weren't attached to `fixed`, and to me one major reason for writing `gear.lossPower` in such a complicated way is exactly that it handles the `support`-case - without adding `support.tau*der(support.phi)`; and I think that should be part of the updated documentation when explaining the check for PowerLoss.

We might also use two PowerSensors (suitably oriented - so good to have that illustrated in Modelica.Mechanics.Rotational) and an Add-block instead of directly writing the PowerLoss-formula - this avoids mixing graphical modeling and textual modeling in one model; and also makes the computed loss more visible.

HansOlsson, 2019-11-11:
I tried to add the sensors as above and the result was:

[image]

```
model LossyGearDemo1
  "Example to show that gear efficiency may lead to stuck motion"
  extends Modelica.Icons.Example;
  Rotational.Components.LossyGear gear(
    ratio=2,
    lossTable=[0, 0.5, 0.5, 0, 0],
    useSupport=true) annotation (Placement(transformation(extent={{-10,0},{10,20}})));
  Rotational.Components.Inertia Inertia1(J=1) annotation (Placement(
        transformation(extent={{-60,0},{-40,20}})));
  Rotational.Components.Inertia Inertia2(
    J=1.5,
    phi(fixed=true, start=0, nominal=0.001),
    w(fixed=true, start=0, nominal=0.01)) annotation (Placement(transformation(extent={{40,0},{
            60,20}})));
  Rotational.Sources.Torque torque1(useSupport=true) annotation (Placement(
        transformation(extent={{-86,0},{-66,20}})));
  Rotational.Sources.Torque torque2(useSupport=true) annotation (Placement(
        transformation(extent={{86,0},{66,20}})));
  Modelica.Blocks.Sources.Sine DriveSine(amplitude=10, freqHz=1)
    annotation (Placement(transformation(extent={{-116,0},{-96,20}})));
  Modelica.Blocks.Sources.Ramp load(
    height=5,
    duration=2,
    offset=-10) annotation (Placement(transformation(extent={{116,0},{96,20}})));
  Rotational.Components.Fixed fixed annotation (Placement(transformation(
          extent={{-10,-70},{10,-50}})));
  Sensors.PowerSensor powerSensor
    annotation (Placement(transformation(extent={{-34,0},{-14,20}})));
  Sensors.PowerSensor powerSensor1
    annotation (Placement(transformation(extent={{16,0},{36,20}})));
  Blocks.Math.Add powerLoss(k2=-1) annotation (Placement(transformation(
        extent={{-10,10},{10,-10}},
        rotation=0,
        origin={50,-30})));
equation 
  connect(torque1.flange, Inertia1.flange_a)
    annotation (Line(points={{-66,10},{-60,10}}));
  connect(torque2.flange, Inertia2.flange_b)
    annotation (Line(points={{66,10},{60,10}}));
  connect(DriveSine.y, torque1.tau)
    annotation (Line(points={{-95,10},{-88,10}}, color={0,0,127}));
  connect(load.y, torque2.tau)
    annotation (Line(points={{95,10},{88,10}}, color={0,0,127}));
  connect(fixed.flange, gear.support)
    annotation (Line(points={{0,-60},{0,0}}));
  connect(fixed.flange, torque1.support)
    annotation (Line(points={{0,-60},{-76,-60},{-76,0}}));
  connect(fixed.flange, torque2.support)
    annotation (Line(points={{0,-60},{76,-60},{76,0}}));
  connect(gear.flange_a, powerSensor.flange_b)
    annotation (Line(points={{-10,10},{-14,10}}, color={0,0,0}));
  connect(Inertia1.flange_b, powerSensor.flange_a)
    annotation (Line(points={{-40,10},{-34,10}}, color={0,0,0}));
  connect(gear.flange_b, powerSensor1.flange_a)
    annotation (Line(points={{10,10},{16,10}}, color={0,0,0}));
  connect(powerSensor1.flange_b, Inertia2.flange_a)
    annotation (Line(points={{36,10},{40,10}}, color={0,0,0}));
  connect(powerSensor.power, powerLoss.u1)
    annotation (Line(points={{-32,-1},{-32,-36},{38,-36}}, color={0,0,127}));
  connect(powerSensor1.power, powerLoss.u2)
    annotation (Line(points={{18,-1},{18,-24},{38,-24}}, color={0,0,127}));
  annotation (Documentation(info="<html>
<p>
This model contains two inertias which are connected by an ideal
gear where the friction between the teeth of the gear is modeled in
a physical meaningful way (friction may lead to stuck mode which
locks the motion of the gear). The friction is defined by an
efficiency factor (= 0.5) for forward and backward driving condition leading
to a torque dependent friction loss. Simulate for about 0.5 seconds.
The friction in the gear will take all modes
(forward and backward rolling, as well as stuck).
</p>
<p>
You may plot:
</p>
<pre>
Inertia1.w,
Inertia2.w : angular velocities of inertias
powerLoss.y  : power lost in the gear
gear.mode  :  1 = forward rolling
              0 = stuck (w=0)
             -1 = backward rolling
</pre>
</html>"),
       experiment(StopTime=0.5, Interval=0.001),
    Diagram(coordinateSystem(extent={{-120,-100},{120,100}})),
    Icon(coordinateSystem(extent={{-120,-100},{120,100}})));
end LossyGearDemo1;
```
Would this be ok as a PR? (And similarly for LossyGearDemo2?)

MartinOtter, 2019-11-13:
> Both of the given formulas are fairly simple so I don't know which you mean "simple loss-power formula"; and we are as far as I know computing all variables.

This is the code to compute tau_loss and I consider this not a simple formula:
```
  ideal = Modelica.Math.Matrices.isEqual(
        lossTable,
        [0, 1, 1, 0, 0],
        Modelica.Constants.eps);

  if ideal then
    interpolation_result = [1, 1, 0, 0];
    eta_mf1 = 1;
    eta_mf2 = 1;
    tau_bf1 = 0;
    tau_bf2 = 0;
  else
    interpolation_result = [
      Modelica.Math.Vectors.interpolate(lossTable[:,1], lossTable[:,2], noEvent(abs(w_a)), 1),
      Modelica.Math.Vectors.interpolate(lossTable[:,1], lossTable[:,3], noEvent(abs(w_a)), 1),
      Modelica.Math.Vectors.interpolate(lossTable[:,1], lossTable[:,4], noEvent(abs(w_a)), 1),
      Modelica.Math.Vectors.interpolate(lossTable[:,1], lossTable[:,5], noEvent(abs(w_a)), 1)];
    eta_mf1 = interpolation_result[1, 1];
    eta_mf2 = interpolation_result[1, 2];
    tau_bf1 = noEvent(abs(interpolation_result[1, 3]));
    tau_bf2 = noEvent(abs(interpolation_result[1, 4]));
  end if;

  if Modelica.Math.isEqual(
          eta_mf1,
          1.0,
          Modelica.Constants.eps) and Modelica.Math.isEqual(
          eta_mf2,
          1.0,
          Modelica.Constants.eps) then
    // For eta_mf1=eta_mf2=1 the given bearing friction can not be
    // separated into a part on side A or B, so it is done arbitrarily.
    tau_bf_a = tau_bf1/2;
  else
    //calculate tau_bf_a from the following equations
    //tau_bf1 = eta_mf1*tau_bf_a + tau_bf_b / ratio
    //tau_bf2 = 1/eta_mf2*tau_bf_a + tau_bf_b / ratio
    tau_bf_a = (tau_bf1 - tau_bf2)/(eta_mf1 - 1.0/eta_mf2);
  end if;

  phi_a = flange_a.phi - phi_support;
  phi_b = flange_b.phi - phi_support;
  phi_a = ratio*phi_b;

  // Torque balance (no inertias)
  0 = flange_b.tau + ratio*(flange_a.tau - tauLoss);

  // Speed for friction element
  w_a = der(phi_a);
  a_a = der(w_a);

  // Determine driving side
  //assuming positive omega
  tau_eta_p = flange_a.tau - tau_bf_a_0;
  //assuming negative omega
  tau_eta_m = flange_a.tau + tau_bf_a_0;

  // assuming w>=0, take value at w=0 to decide rolling/stuck mode
  quadrant1_p = (1 - eta_mf1_0)*flange_a.tau + tau_bf1_0;
  quadrant2_p = (1 - 1/eta_mf2_0)*flange_a.tau + tau_bf2_0;
  tauLossMax_p = if noEvent(tau_eta_p > 0) then quadrant1_p else 
    quadrant2_p;

  // assuming w<=0, take value at w=0 to decide rolling/stuck mode
  quadrant4_m = (1 - 1/eta_mf2_0)*flange_a.tau - tau_bf2_0;
  quadrant3_m = (1 - eta_mf1_0)*flange_a.tau - tau_bf1_0;
  tauLossMin_m = if noEvent(tau_eta_m > 0) then quadrant4_m else 
    quadrant3_m;

  quadrant1 = (1 - eta_mf1)*flange_a.tau + tau_bf1;
  quadrant2 = (1 - 1/eta_mf2)*flange_a.tau + tau_bf2;
  quadrant4 = (1 - 1/eta_mf2)*flange_a.tau - tau_bf2;
  quadrant3 = (1 - eta_mf1)*flange_a.tau - tau_bf1;

  //tau eta: only for determination of driving side for calculation of tauloss
  tau_eta = if ideal then flange_a.tau else (if locked then flange_a.tau
     else (if (startForward or pre(mode) == Forward) then flange_a.tau -
    tau_bf_a else flange_a.tau + tau_bf_a));

  // Torque Losses
  tau_etaPos = tau_eta >= 0;
  tau_aPos = tau_etaPos;
  tauLossMax = if tau_etaPos then quadrant1 else quadrant2;
  tauLossMin = if tau_etaPos then quadrant4 else quadrant3;

  // Determine rolling/stuck mode when w_rel = 0
  startForward = pre(mode) == Stuck and sa > tauLossMax_p/unitTorque or 
    initial() and w_a > 0;
  startBackward = pre(mode) == Stuck and sa < tauLossMin_m/unitTorque or 
    initial() and w_a < 0;
  locked = not (ideal or pre(mode) == Forward or startForward or pre(mode)
     == Backward or startBackward);

  /* Parameterized curve description a_a = f1(sa), tauLoss = f2(sa)
       In comparison to Modelica.Mechanics.Rotational.FrictionBase it is possible
       to simplify the following expression as mode == Stuck is assured in case
       of startForward or startBackward */
  tauLoss = if ideal then 0 else (if locked then sa*unitTorque else (if (
    startForward or pre(mode) == Forward) then tauLossMax else tauLossMin));
```
Once tau_loss is computed, the loss power is simply computed by `lossPower = tauLoss*w_a;`. If there is a bug in the code for tau_loss, then the lossPower will be wrong. To check the correctness, the loss power is computed in the examples by the sum of the powers over all connectors. This computation is only possible if all variables, such as, tauLoss are computed.

MartinOtter, 2019-11-13:
> 
> 
> I tried to add the sensors as above and the result was:
> 
When improving the model in this respect, the difference with the internally computed loss power should be added and explaining that this should be close to zero.

beutlich, 2020-01-12:
@MartinOtter I am not sure why you removed your assignment. On 2019-10-18 I assigned you (https://github.com/modelica/ModelicaStandardLibrary/issues/2934#issuecomment-543565484) after it somehow was decided that only the documentation needs to be updated. Afterwards, a lot of models have been posted w/o any feedback.

@MartinOtter @HansOlsson What is the state of this issue after all?

HansOlsson, 2020-01-13:
> @MartinOtter I am not sure why you removed your assignment. On 2019-10-18 I assigned you (#2934 (comment)) after it somehow was decided that only the documentation needs to be updated. Afterwards, a lot of models have been posted w/o any feedback.
> @MartinOtter @HansOlsson What is the state of this issue after all?

I don't know. 

To me the current model is unclear and at least the documentation needs to be updated (if it takes a specialist days to understand why some variables exist that is a clear sign), but I'm not an officer for this library and I don't want to step on any toes by doing that.

tobolar, 2020-01-28:
Following good practice, the PowerLoss should be better generated graphically by means of two power sensors (`Modelica.Mechanics.Rotational.Sensors.PowerSensor`) on gear input and output, and an Add block for subtraction of those two power signals.

PR #3373 (2020-02-01): Documentation of LossyGearDemos improved (fixes #2934)
Pull request to fix #2934
commit 84a8a4bd (2020-02-01): Documentation of LossyGearDemos improved (fixes #2934)
commit e44435ab (2020-02-01): Merge pull request #3373 from modelica/ImproveDocuLossyGearDemos

Documentation of LossyGearDemos improved (fixes #2934)
closed 2020-02-01 by beutlich