modelica/ModelicaStandardLibrary #2936: Illegal usage of conditional component in Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMEE_DOL

perost, 2019-05-15:
The `Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMEE_DOL` model contains the declaration:
```modelica
Modelica.SIunits.Current irRMS = sqrt(smee.ir[1]^2+smee.ir[2]^2)/sqrt(2);
```
`ir` inside `smee` is a conditional component though, so using it in this context is not allowed since conditional components may only be modified and/or used in connections.

christiankral, 2019-05-16:
In this example it makes only sense to have the damper cage present. Otherwise the machine cannot start. So the parameter `useDamperCage` will be fixed so that the cage cannot be disabled.

sjoelund, 2019-05-21:
I think the following might work (didn't check it):
```mo
  Modelica.Blocks.Interfaces.RealOutput ir[2];
  Modelica.SIunits.Current irRMS = sqrt(ir[1]^2+ir[2]^2)/sqrt(2);
equation
  connect(ir, smee.ir);
```

christiankral, 2019-05-26:
This solution is not working. I tested it in Dymola and OpenModelica.

My conclusions: In case of having the cage not present, the connection 
```modelica
connect(ir, smee.ir); 
```
is removed and there are then no equations for the determineation of `ir[2]`...

sjoelund, 2019-05-26:
Oh yeah, I guess both need to have the same if condition...

AHaumer, 2019-05-27:
In Modelica.Electrical.Machines.BasicMachines.SynchronousInductionMachines.* and Modelica.Magnetic.FundamentalWave.BasicMachines.SynchronousInductionMachines.* we have the same problem: The DamperCage is continitional. Nevertheless we want to have an information about the DamperCage currents (if the DamperCage is present), and we need them for initialization (if the DamperCage is present). The solution with connect is fine and should work. So we know the components of the DamperCage current.
The problem is to provide the RMS of the DamperCage current only if the the DamperCage is present. 
I doubt that the conditional definition (and calculation) of iRMS works (since conditional variables are used).
@christiankral : We could define iRMS in the conditional DamperCage), so we could connect a conditional iRMS to to damperCage.iRMS (preferrably in the machine model).

sjoelund, 2019-05-27:
```mo
  Modelica.Blocks.Interfaces.RealOutput ir[2] if useDamperCage;
  Modelica.SIunits.Current irRMS = sqrt(ir[1]^2+ir[2]^2)/sqrt(2) if useDamperCage;
equation
  connect(ir, smee.ir);
```
You are allowed to use bindings for `irRMS`. So both being conditional should work. In SMEE_DOL the source code says the cage is always present though, and should work without that if-condition.

The problem is if you want to use that irRMS anywhere (requires other elements to use the same if condition).

henrikt-ma, 2019-05-27:
Are you really allowed to use the conditional `ir` in the binding equation of another component (conditional or not)?

beutlich, 2019-05-27:
I doubt that @sjoelund's proposal is leagal since

> Conditional components may only be modified or used in connections.

sjoelund, 2019-05-28:
So yes. Either my original proposal (which breaks the model if the examples are extended to not use the damper cage anymore). Or add an additional connector in base model with `if not useDamperCage` that sets it to some predefined value...

AHaumer, 2019-06-04:
Just a question: As far as I see we should apply the same solution to maint/3.2.3 - right?

AHaumer, 2019-06-04:
See PR #2958 (issue #2936 same changes for maint/3.2.2 as PR #2956 for master)

beutlich, 2019-06-12:
Resolved in master and maint/3.2.3 branches.

PR #2956 (2019-06-04): Illegal usage of conditional component in Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.SMEE_DOL
resolves #2936
closed 2019-06-12 by beutlich