modelica/ModelicaStandardLibrary #2940: Translational driving resistance

AHaumer, 2019-05-19:
It would be nice to have a translational driving resistance at hand, containing drag resistance (dependent on quare of speed), rolling resistance and gravitational resistance (depending on inclination).

beutlich, 2019-05-20:
See also #1873.

tobolar, 2019-05-20:
As such component is applicable especially for ground vehicles, maybe you can explain the background of necessity to implement it within a standard library.
Actually, there are yet some components facilitating the assembly of such part: `QuadraticSpeedDependentForce` for drag and `SignForce` for the rolling resistance:
[image]

(If one considers just small slope angles, the lower part would even simplify.)

HansOlsson, 2019-05-20:
@tobolar Good point, and I would only like to add that forceRolling could also be modelled as a Brake, which both allows for locking at small speeds - and allows for variable normal force due to the slope (for large slopes).

AHaumer, 2019-05-20:
First, sorry for mixing um #2935 and #2940. This one corresponds with #2940!
@beutlich How is that correlated to #1873?
@tobolar forceDrag and forceSlope are correct, forceRolling is not since normal force varies with inclination (see comment of @HansOlsson).
Yes this component is needed for vehicles. I do a lot of modeling of such vehicles with students in my courses and I'm tired of implementing it over and over again - better use a standardized component from MSL.
@HansOlsson you're right, this would allow to implement stick force. From a didactical point of view, the implementation using equations is much clearer (Why is a brake in a driving resistance?). Additionally, locking force has to be specified (which is not the case for simple examples).

AHaumer, 2019-05-20:
@HansOlsson 
If rolling resistance coefficient cR varies due to varying road condition, it is much easer to take care of that using equations than using the brake component.

dietmarw, 2019-05-20:
Wouldn't be the VehicleInterfaces be the better place to have such components? If we start adding all components that are nice to have because of domain specific applications then we we get a problem. Of course it is convenient to have everything inside the MSL but with that reasoning I could start putting Power Systems components into MSL instead of using a application library (that uses MSL). And that should not be the way to get a proper Modelica **Standard** Library.

AHaumer, 2019-05-20:
@dietmarw 
For simple examples, I don't need VehicleInterfaces - so why put it there and open VehicleInterfaces for just one component? Driving resistance is a rather basic effect for translational movement.

HansOlsson, 2019-05-20:
> Wouldn't be the VehicleInterfaces be the better place to have such components? If we start adding all components that are nice to have because of domain specific applications then we we get a problem.

Minor note: I believe it should be in another library as I thought VechicleInterfaces was for the interfaces, not the actual implementations. 

However, I agree that there is a risk of feature creep of MSL - but then we a clear guideline and use it e.g., also for #2938

tobolar, 2019-05-20:
@dietmarw That's exactly what I want to point out. Where are the (self defined) limits for MSL?
@AHaumer For forceRolling you are right.

AHaumer, 2019-05-20:
@tobolar Equation based or built from components - depends which implemntation is easier to understand and easier to maintain
@HansOlsson "feature creep" - do you think we should NOT implement new features?
Offering more useful models will attract more users ...

HansOlsson, 2019-05-20:
> @HansOlsson "feature creep" - do you think we should NOT implement new features?

Clearly it is good to implement this new model. 

The question is whether it should be in MSL or not - and I merely stated that we should have **clear guidelines** for that; nothing about this particular model.

I think the Buildings-library has shown us that good Modelica contents doesn't need to be in MSL (or even handled by MA).

AHaumer, 2019-05-20:
If we're talking about a whole library for special applications, you're right. Could (should?) be a separate applications library, preferably maintained by MA.
What's about single models? Should we go through the whole MSL and pull out all models that are not "basic enough" into a collection of "application models"?
I'd prefer to have such basic models in the MSL at hand in every Modelica tool.

AHaumer, 2019-05-20:
VehicleInterfaces is the wrong place. It's not "interfaces". Furthermore, VehicleInterfaces can be switched between 1d and 3d - this model is purely 1d.

beutlich, 2019-05-20:
> @beutlich How is that correlated to #1873?

#1873 is on deired mechanical components and thus could serve as an umbrella ticket for #2940, since the driving resistance can be seen as a desired mechanical component.

> I think the Buildings-library has shown us that good Modelica contents doesn't need to be in MSL (or even handled by MA).

On the other side, the Building library implements lots of core features (or features independent of the Buildings domain) which could go to MSL or a separate feature library.

> The question is whether it should be in MSL or not - and I merely stated that we should have clear guidelines for that; nothing about this particular model.

I do not see a problem adding this model to the MSL even though it looks like an application component from automotive power transmission. As a compromise, we could add it to Modelica.Mechanics.Translational.Examples.Utilities and use it in a new example model.

tobolar, 2019-05-22:
> The question is whether it should be in MSL or not - and I merely stated that we should have clear guidelines for that; nothing about this particular model.

I fully agree.
If possible, it shall be clear (a) why we introduce a new component _which addresses just a particular application_ and (b) _what level of detail_ shall be implemented. It is of importance too, as demonstrated in the following.
1. There is a `Mechanics.Rotational.Components.LossyGear` modelling losses in a gearbox. In #1873, there is suggested to have a simpler model of a gearbox with constant efficiency only - in order to facititate the parametrization.
2. Concerning driving resistance, PR #2942 suggests some model being quite generic and thus requiring a couple of parameters and inputs being provided. But regarding point 1 we could expect there will come a request in the future to provide a simpler model as well - e.g. provided for small slopes only - or a more complicated model concerning e.g. tyre air resistance.

AHaumer, 2019-06-02:
see PR #2954

PR #2963 (2020-01-13): Implement simple vehicle model
implement simple vehicle model (resolve #2940 / alternative PR to #2954)
closed 2020-01-13 by beutlich