modelica/ModelicaStandardLibrary #2941: Initialization of bearingFriction.sa in RobotR3.oneAxis

henrikt-ma, 2019-05-20:
In #1741, the first example of models that are problematic for regression testing pointed to the initialization of `axis.gear.bearingFriction.sa`.  This has always been a bit worrying, but so far SystemModeler hasn't really been aware of the problem, just carrying out the fixed point iteration and being happy to find a solution.

Now, we'd like to do better, and when doing our refined analysis of initialization problems, the problem with `oneAxis` and `fullRobot` stand out as one out of a very small number of issues among all the MSL examples.  Therefore, I'm opening this separate issue to get a more focused discussion than what is possible in the overview ticket #1741.

In detail: How is a tool supposed to initialize `axis.gear.bearingFriction.sa`, not knowing the value of `axis.gear.bearingFriction.locked`?

beutlich, 2019-10-06:
@MartinOtter @tobolar The issue #1741 from @frenkelj on FullRobot initialization is open for more than four years with a detailed analysis provided. Please let me know when you think you can provide a fix for it!

MartinOtter, 2019-10-16:
> In detail: How is a tool supposed to initialize axis.gear.bearingFriction.sa, not knowing the value of axis.gear.bearingFriction.locked? 

locked is declared as:
```
Boolean locked(start=false)
```
Therefore, the start value is well-defined (= false) and the iterative algorithm has to start with this value.

henrikt-ma, 2019-10-17:
How does this answer the following statement in #1741?
> The problem is that for initialization `locked` needs to be true and in this case `sa` is not incident in the equation.

MartinOtter, 2019-10-18:
Sorry, your answer is too short and I do not get your argument. Please, give more explanations (it is standard, that you start in a nonlinear equation system with a guess value and then iterate until convergence is reached. There might be multiple solutions and typically the solution closes to the guess value is selected, but this is not guaranteed).

henrikt-ma, 2019-10-23:
The point of this issue is that the way this model tries to initialize is not standard.

I think it has been established that Modelica models can have equation systems that pass through numerical singularities, and this is generally not a big problem when solving the equations if it only happens during integration at isolated points in time.  It is a bigger problem if numeric singularity happens during initialization, as `start` attributes are just guess values, not the most recent point on an otherwise well-defined trajectory.

On the other hand, structural singularity is expected to be detected during symbolic processing, and be fixed in the model rather than coped with during simulation.

What we have here is a linear system with a structural structural singularity conditioned on variables solved during simulation, variables that belong to the same mixed system.

All one needs to do to make the example collapse is to set the initial acceleration `axis.const.k` to anything but zero.  One way of avoiding the risk of collapse would be to use a source with constant variability, as proposed in https://github.com/modelica/ModelicaStandardLibrary/pull/3112, but there are other ways as well, including making a `final` modification of `axis.const.k` with a constant value in the top level model.  Protecting against the collapse in such ways, however, only mitigates the consequences of the structural singularity.  The model's initialization would still be more well-posed if the (conditioned) structural singularity was removed altogether.

Unfortunately, all I can do at the moment is clarify the problem.  I don't have a solution to it.