modelica/ModelicaStandardLibrary #2944: Convert the obsolete types FirstOrderTemperaturCoefficient and SecondOrderTemperaturCoefficient

beutlich, 2019-05-20:
Reminder to convert the obsolete types FirstOrderTemperaturCoefficient and SecondOrderTemperaturCoefficient

christiankral, 2019-06-16:
I checked the master branch and found 

```
type LinearTemperatureCoefficient = Real(final quantity = "LinearTemperatureCoefficient", final unit="1/K");

type QuadraticTemperatureCoefficient = Real(final quantity = "QuadraticTemperatureCoefficient", final unit="1/K2");

type LinearTemperatureCoefficientResistance = Real (
  final quantity="LinearTemperatureCoefficientResistance",
  final unit="Ohm/K") "First Order Temperature Coefficient";

type QuadraticTemperatureCoefficientResistance = Real (
  final quantity="QuadraticTemperatureCoefficientResistance",
  final unit="Ohm/K2") "Second Order Temperature Coefficient";
```

@beutlich Is there anything left to do?

beutlich, 2019-06-16:
Yes, this issue is on the conversion (and removal) of the misspelled types FirstOrderTemperaturCoefficient and SecondOrderTemperaturCoefficient (note, the missing **e**). I created #2930 for this, but you questioned if they are in the right package: NonSIunits or SIunits. #2930 was straight forward, but now I longer know, how to convert these two misspelled types.

christiankral, 2019-06-17:
Do I understand correctly:

- [x] The misspelled units `FirstOrderTemperaturCoefficient` and `SecondOrderTemperaturCoefficient` are fixed. 
- [x] The non SI units are replaced by SI units, as `degC` is replaced by `K` and `degC2` is replaced by `K2`.
- [x] The two units are moved from `NonSIunits` to `SIunits`. 
- [ ] We are currently only missing a conversion script but the question is what is the syntax to convert the two types properly? Is that the missing task? So
```
convertClass("Modelica.SIunits.Conversions.NonSIunits.FirstOrderTemperaturCoefficient",
              "Modelica.SIunits.LinearTemperatureCoefficientResistance");
``` 
is not the right solution (for one of the two)?

beutlich, 2019-06-18:
>     * Is that the missing task?

See https://github.com/modelica/ModelicaStandardLibrary/pull/2930/files#diff-a8803965a30a4c681f147defe8b74ab7. Beside the conversion we also need to remove the misspelled types and require a conversion test model. Can you please revisit #2930 again.

PR #2981 (2019-06-19): Convert FirstOrderTemperaturCoefficient and SecondOrderTemperaturCoefficient
Fix #2944
closed 2019-06-19 by beutlich