modelica/ModelicaStandardLibrary #2945: Removing usage of Clock as a class name in MSL

Daiwen, 2019-05-21:
I would like to propose that we modify the MSL to not use Clock as a class name. This would resolve some ambiguities for the user, and make it possible for the language design group to make Clock a keyword of the Modelica language.

Conversion rules would make that change somewhat easier.

beutlich, 2019-05-22:
@Daiwen Should https://github.com/modelica/ModelicaSpecification/issues/2280 be reopened?

henrikt-ma, 2019-05-22:
Before reopening https://github.com/modelica/ModelicaSpecification/issues/2280, I suggest a small scope creep of this issue to cover all introduced `Clock` identifiers in the MSL.  As long as the MSL has components called `Clock`, there's still nothing we can do about it's keyword status.

Daiwen, 2019-05-23:
I hadn't seen the case of component names but Henrik is right we can't reopen  modelica/ModelicaSpecification#2280 until there are still occurrences of `Clock` whatever they maybe.
I can open a new issue if you'd prefer me doing it that way.

beutlich, 2019-05-23:
OK, I got it.

PR #2947 (2019-06-12): Convert Clock
Rename
* Modelica.Blocks.Sources.**Clock** to Modelica.Blocks.Sources.ContinuousClock
* Modelica.Electrical.Digital.Sources.**Clock** to Modelica.Electrical.Digital.Sources.DigitalClock

Close #2945.

@Daiwen FYI
closed 2019-06-12 by beutlich