modelica/ModelicaStandardLibrary #2952: CODATA 2018 Update (Big G)

MrMatt2532, 2019-06-02:
Big G was updated with the latest CODATA 2018 release. This was missed in the recent 2017 CODATA/SI change.

See NIST reference here: [Newtonian Constant of Gravitation](https://physics.nist.gov/cgi-bin/cuu/Value?bg#mid)

HansOlsson, 2019-06-03:
> Big G was updated with the latest CODATA 2018 release. This was missed in the recent 2017 CODATA/SI change.

As far as I understand it wasn't "missed" - it was deliberately skipped as "G" isn't used in the redefinition of the SI-system. 

However, I don't see a good reference for the new value - I would assume codata 2018 will be published similarly as codata 2014 - but that hasn't happened yet as far as I can tell, and https://physics.nist.gov/ seems to regularly go down.

MrMatt2532, 2019-06-04:
Yes, I only meant it was "missed" in the sense that this new recommended value appears to have come into effect on the same date as the new SI changes (May 20th). With that said, yes I agree it looks like the published reference isn't available yet. Likely in the coming months...

PR #3251 (2019-12-05): Update constant G according to CODATA 2018 value
Fix #2952.
closed 2019-12-05 by beutlich