modelica/ModelicaStandardLibrary #2960: import Modelica.SIunits

AHaumer, 2019-06-06:
In my opinion, the usage of `import SI=Modelica.SIunits;` at some place in a sublibrary to shorten parameter definitions is really bad style:
- hard to read: Where is the import statement located?
- If you copy a model out of the MSL, the import gets broken.
Much better style would be: `import Modelica.SIunits.*;` _in every model_. 
- easy to read: The import statement is in the model I'm looking at.
- If you copy a model out of the MSL, the import goes with the model, nothing gets broken. 

The question is:
Should we go through the whole MSL and change the style (resp. location) of such import statements?

christiankral, 2019-06-06:
I think this is a **very good idea!** I am in favor of improving the quality of the Modelica Standard Library that way!

AHaumer, 2019-06-06:
Compare 3 styles:
1. `parameter Modelica.SIunits.Mass m=100 "Total mass";`
Clear but not elegant.
2. `parameter SI.Mass m=100 "Totoal mass";`
`import SI=Modelica.SIunits` elsewhere
Neither Clear nor elegant.
3. `parameter Mass m=100 "Totoal mass";`
`import Modelica.SIunits.*` just above
Clear and elegant.

dietmarw, 2019-06-07:
Sorry, but **NEVER EVER** use wildcard imports. We even write in the specs to not use them as they are dangerous since there is a huge danger of variable collision. So for imports only ever use implicit imports. They are simply bad programming style. 
As for `import SI=Modelica.SIunits ` this is something that is down to taste. I use this, and like this, extensively together with `import C=Modelica.Contstants`. So definitely don't start changing this in the MSL. I find it much easier on the eye to not have the whole `Modelica.SIunits.` prefix all over the place.

AHaumer, 2019-06-07:
What about copy a model out of the MSL for applying own changes?

HansOlsson, 2019-06-10:
> What about copy a model out of the MSL for applying own changes?

I see three possible solutions:

- Use a tool that automatically handles this when copying. 
- Always use full path, e.g., Modelica.SIunits.Mass
- Have `import SI=Modelica.SIunits;` in every model. I find that solution tedious.

dietmarw, 2019-06-11:
The solution would definitely be on the tool side in this case. And Dymola does this already by recognising the imported definitions and replacing them by full qualified class names upon duplication. So it really is a non-issue. And if it is then it is a missing tool issue. I know that OpenModelica for example doesn't even recognise relative name lookup on duplication at the moment. But they are working on it and the recognition of imports would just be another such feature that needs then to be worked on by the tool (@adrpo FYI).

AHaumer, 2019-06-12:
ok you're right

closed 2019-06-12 by AHaumer

beutlich, 2019-06-12:
> Compare 3 styles:
>
>     1. `parameter Modelica.SIunits.Mass m=100 "Total mass";`
>        Clear but not elegant.
>
>     2. `parameter SI.Mass m=100 "Totoal mass";`
>        `import SI=Modelica.SIunits` [...]

The recommendation is to use unnamed import statements, i.e.,

```mo
parameter SIunits.Mass m=100 "Totoal mass";
import Modelica.SIunits;
```