modelica/ModelicaStandardLibrary #2961: feature creep

AHaumer, 2019-06-06:
I hate the discussions about "feature creep" - we have them again and again ...

I have a rather clear opinion, we have 2 cases:

1.  A whole new library, maybe in a new domain, maybe for a type of application.
We should discuss whether to include this library as a new sublibrary in the MSL, or to have a new companion library of the MSL - under the same rules of quality assurance.

2. New models or enhancement of models - part of an existing sublibrary of the MSL.
In most cases discussion should be short to decide whether this is a model of broad interest (based on textbook knowledge -> include in the MSL) or rather a model for a specific application. Should we collect such models, and if yes, where? 

@christiankral and I and a some other people are teaching a lot utilizing Modelica. This is perfect advertisement for new/future users. Show students how easy they can tackle a problem utilizing Modelica and the're convinced to use Modelica in their professional future, too. 

The more models based on textbook knowledge for applications of broad interest we have in the MSL, the easier we will attract new users.

beutlich, 2019-08-04:
This always has been discussed controversially in the past. We have seen good and bad examples with the inclusion of new sub-libraries in the MSL. For that reason people tend to be either inclusive or exclusive. Personally, I would be more than happy if we really manage to add the main new feature libraries [Clocked](https://github.com/modelica/ModelicaStandardLibrary/issues/2862), [Electrical.Batteries](https://github.com/modelica/ModelicaStandardLibrary/issues/2957) and [Mechanical.MultiBody.FlexibleParts](https://github.com/modelica/ModelicaStandardLibrary/issues/2938) in MSL v4.0.0.

beutlich, 2019-11-02:
Seems to have been clarified for upcoming MSL v4.0.0.

closed 2019-11-02 by beutlich