modelica/ModelicaStandardLibrary #2964: Triac model

AHaumer, 2019-06-08:
I do not understand the `Modelica.Electrical.Analog.Ideal.IdealTriac`:

Both thyristors are fired by the same boolean signal?
Shouldn't they get fired by 2 boolean signals with 180° (1/2 period) delay in between?
Does the model rely on the fact that the positive thyristor that is fired during the negative halfwave, too, wont start conducting since voltage is in reverse direction, and for the negative thyristor the other way round?

What are the resistor (Rdis) and capacitor (Cdis) good for? They can cause a rather high leakage current if the triac is in blocking condition, and a model without these would work without problems.

Maybe @christophclauss and @kristinmajetta could clarify this issue. thanks.

AHaumer, 2019-06-10:
To describe the triac on an abstract level, the fire signal has to have 3 values:
fire = 0 stay in blocked condition (no one of the thyristors fired).
fire = +1 fire the "positive" thyristor, current starts to flow in positive direction if voltage is positive. 
fire = -1 fire the "negative" thyristor, current starts to flow in negative direction if voltage is negative. 
The triac returns to blocked condition when current gets zero.

christiankral, 2019-07-04:
The triac model `Modelica.Electrical.Analog.Ideal.IdealTriac` is on my opinion not an **ideal** model as it also has a resistor and a capacitor parallel connected to the thyristors to eliminate disturbances. 

[image]

As this capacitor may interfere with other components of an electric circuit it rather were a `TriacWithDisturbanceElimination` than an `IdealTriac`. I thus propose to **rename** the `IdealTriac`. 

@christophclauss Do you agree to rename `IdealTriac` to, e.g. `TriacWithDisturbanceElimination`. 

Additionally I propose to move the Triac model proposed in #2970 to `Modelica.Electrical.Analog.Ideal.IdealTriac`, as this is an idea triac model:

[image]

christophclauss, 2019-08-04:
The considerations sound o.k. Nevertheless, I need the original source on which the actual model is based on.  Possibly we should offer an additional model, an advanced one. It seems to take too much time to at the moment, therefor I'd like postpone this issue after MSL4.0.0

beutlich, 2019-08-06:
>  I need the original source

Is there a chance you can get the original source from @kristinmajetta ?

christiankral, 2020-02-15:
I see the MSL 4.0.0 tag was removed some time ago.

Yet, @AHaumer how are we proceeding here? If were are not doing anything before MSL4.0.0 is released we will still have a non-ideal component named `IdealTriac`.

One simple solution could be:

- Rename `Modelica.Electrical.Analog.Ideal.IdealTriac` to `Modelica.Electrical.Analog.Ideal.TriacWithDisturbanceElimination`
- Make a note in this model that an ideal model is included in `Modelica.Electrical.PowerConverters.ACAC.SinglePhaseTriac` through #2970