modelica/ModelicaStandardLibrary #2968: Missing reference results

maltelenz, 2019-06-12:
All of the below concerns version 3.2.3.

These models do not seem to have any reference results in the reference results repository (or traces that one was attempted). They all have experiment annotations though:

- ModelicaTest.Math.TestMatricesExamplesSolveLinearEquations
- ModelicaTest.Tables.CombiTimeTable.Test85
- ModelicaTest.Tables.CombiTimeTable.Test86
- ModelicaTest.Utilities.TestExamplesCalculator
- Modelica.Electrical.Machines.Examples.ControlledDCDrives.SpeedControlledDCPM
- Modelica.Magnetic.QuasiStatic.FluxTubes.Examples.Leakage.GeneralLeakage
- Modelica.Mechanics.Rotational.Examples.EddyCurrentBrake
- Modelica.Mechanics.Translational.Examples.EddyCurrentBrake

These models seem to be missing the csv file:

- ModelicaTest.Fluid.TestComponents.Sensors.TestTemperature1
- ModelicaTest.Blocks.LimitersHomotopy
- Modelica.Blocks.Examples.Rectifier12pulseFFT
- Modelica.Blocks.Examples.Rectifier6pulseFFT
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.HysteresisModelComparison
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.InductorWithHysteresis
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1
- Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis2

christiankral, 2019-06-17:
@GallLeo Can you more or less create the test results automatically by reading the model experiment annotation in an automated way, if I provide the proper list of variables in a file named `comparisonSignals.txt`?

christiankral, 2019-06-20:
Refs https://groups.google.com/forum/#!msg/map-lib/jDbvOWGbQNY/klMtdIgHAgAJ

beutlich, 2020-01-19:
Will be resolved by #3348.

maltelenz, 2020-01-20:
Will #3348 also be merged into the 3.2.3 branch? If it is only for master/4.0.0, I don't agree that it resolved this issue, which is for 3.2.3.

beutlich, 2020-01-20:
> 
> 
> Will #3348 also be merged into the 3.2.3 branch? If it is only for master/4.0.0, I don't agree that it resolves this issue, which is for 3.2.3.

I do not think that anyone is going to repeat the MSL v3.2.3 regression run. My focus is MSL v4.0.0 now. Since #3348 is pretty time-consuming (with all the new models, renamed models, renamed signals or even non-transient simulation results (see e.g. #3349)) and since this issue it is not a real bug of MSL v3.2.3 (just a shortcoming of the previous regression runs), it will be good enough if we fix it for MSL v4.0.0.

maltelenz, 2020-01-20:
Missing the reference results means no tools can verify they correctly handle these models.

Focusing on 4.0.0 is certainly a valid way to prioritize, but it still doesn't fix this problem for 3.2.3, and therefore doesn't fix this issue.

beutlich, 2020-01-20:
This is a process issue since previously the introduced new example models were somehow forgotten. For example, I reminded @GallLeo to update the correspondig reference signals in case of new models (see https://github.com/modelica/ModelicaStandardLibrary/pull/2738#issuecomment-429625984 for an example) but it never happened. With #3348 all reference signals are part of the repository and also in responsibility of the library officers. I do hope that this fixes the process in general, whereas it does not fix the MSL v3.2.3 shortcoming you reported. Sorry for that.

beutlich, 2020-02-10:
Meanwhile, CSV files for

* Modelica.Magnetic.FluxTubes.Examples.Hysteresis.HysteresisModelComparison
* Modelica.Magnetic.FluxTubes.Examples.Hysteresis.InductorWithHysteresis
* Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis1
* Modelica.Magnetic.FluxTubes.Examples.Hysteresis.SinglePhaseTransformerWithHysteresis2

are available by https://github.com/modelica/MAP-LIB_ReferenceResults/commit/aa90b8158c02c8f2d5ad6e12ddc05a329aace596.

beutlich, 2020-03-16:
Should all be fixed by the newly created reference results of upcoming MSL 3.2.3 Build.4: https://github.com/modelica/MAP-LIB_ReferenceResults/commits/v3.2.3%2Bbuild.4

closed 2020-03-16 by beutlich