modelica/ModelicaStandardLibrary #2971: Conversion/Renaming Machines

AHaumer, 2019-06-13:
@christiankral @beutlich @dietmarw @HansOlsson 
For a long time we did wait for a release with conversion script for a bunch of renaming, especially AsynchronousInductionmachine -> InductionMachine and SynchronousInductionmachine -> SynchronousMachine.

I've attached a conversion script, i.e. conversion of user models shouldn't be a problem.
Renaming with a smart tool should fix the instatntions within MSL. 
But what happes with open pull requests? How are these converted?

I kindly ask for your advice how to handle this. 
[ConvertMachines.txt](https://github.com/modelica/ModelicaStandardLibrary/files/3287148/ConvertMachines.txt)

AHaumer, 2019-06-13:
One more question: Are wildcards allowed in the conversion script?

beutlich, 2019-06-13:
> For a long time we did wait for a release with conversion script for a bunch of renaming, especially AsynchronousInductionmachine -> InductionMachine and SynchronousInductionmachine -> SynchronousMachine.

I guess you address #733 with your conversion.

> But what happes with open pull requests? How are these converted?

Manually. For that reason (of depending changes) it is recommended to have not too many PRs open and to review PRs in time.

See also #2914 for a major renaming task.

dietmarw, 2019-06-14:
> One more question: Are wildcards allowed in the conversion script?
If you want to rename all child classes of a parent class then you simply rename the parent class. So that would be the equivalent of the wildcard you are asking about.

Could we close this issue here and continue on the original #733 perhaps to not digress too much?

christiankral, 2019-06-15:
I propose to go one step ahead considering #2973:

Change in [ConvertMachines.txt](https://github.com/modelica/ModelicaStandardLibrary/files/3287148/ConvertMachines.txt):

```
convertClass("Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_DOL_MultiPhase",
             "Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.IMC_DOL_MultiPhase");
convertClass("Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMS_Start_MultiPhase",
             "Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.IMS_Start_MultiPhase");
```

to:

```
convertClass("Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMC_DOL_MultiPhase",
             "Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.IMC_DOL_Polyphase");
convertClass("Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.AIMS_Start_MultiPhase",
             "Modelica.Magnetic.FundamentalWave.Examples.BasicMachines.IMS_Start_Polyphase");
```

beutlich, 2019-06-16:
We can rename example models as we like. There is no need to have conversion scripts for them.

beutlich, 2019-06-22:
I started to work on it. It is getting massive.

beutlich, 2019-06-22:
Closing in favour of #733.

closed 2019-06-22 by beutlich