modelica/ModelicaStandardLibrary #2972: Cubic interpolation moody diagram

dofware, 2019-06-14:
Dear all,
we have an issue about the interpolation cubic spline expression at the following modelica standard fluid library line:

https://github.com/modelica/ModelicaStandardLibrary/blob/e2dafbe029f7171130f46800077251834a40e716/Modelica/Fluid/Pipes.mo#L2930

We are not able to obtain the same "Re" expression as the one implemented. In particular the "c3" coefficient in the expression multipies the "dx^2" term, while to us the final expression should be different.

Could anybody help us about this issue?

Thanks a lot

beutlich, 2019-06-14:
@casella @thorade Can you please have a look at it.

beutlich, 2019-06-21:
@wischhusen I noticed that the aboved mentioned function Modelica.Fluid.Pipes.BaseClasses.WallFriction.PartialWallFriction.massFlowRate_dp.interpolateInRegion2 is a duplicate of Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_Re (previously named CubicInterpolation_DP) and Modelica.Fluid.Pipes.BaseClasses.WallFriction.PartialWallFriction.pressureLoss_m_flow.interpolateInRegion2 of Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_lambda (previously named CubicInterpolation_MFLOW), respectively. Is there any chance we can avoid the (undocumented) protected functions interpolateInRegion2 in favour of the well-documented functions in Fluid.Dissipation.Utilities.Functions.General?

wischhusen, 2019-07-15:
Yes, it should be possible by calling the function from Fluid.Dissipation.Functions.General. But I have to admit, that this particular function is undocumented, too. It seems that we picked it from the Modelica_Fluid package of Summer 2006. I am not sure, who implemented the original code but it must have been from the people that created the predecessor of the Modelica.Fluid library. Eventually, Martin Otter recalls who did the work?

beutlich, 2019-10-06:
Unfortunately nobody is able to recall why it is like this since the first addition of Modelica_Fluid. There is some commit https://github.com/modelica-deprecated/Modelica_Fluid/commit/0e76ac5f2b523786daa2d09016ee462bdc186e30 by @dietmarw where the equation `Re := Re1*(lambda2/lambda2_1)^(1 + dx*(c2 + dx*c3));` was added to the StreamConnector branch. Having no access to http://groups.google.com/group/ModelicaTask-Fluid/browse_thread/thread/20dfbadd5bb6c089, to me it looks that there have been parallel developments of Modelica_Fluid where the StreamConnector branch finally was the one to be merged to MSL.

> We are not able to obtain the same "Re" expression as the one implemented. In particular the "c3" coefficient in the expression multipies the "dx^2" term, while to us the final expression should be different.

@dofware Can you please elaborate in more detail which expression do you obtain? Thank you!

beutlich, 2019-10-28:
I tested it with the following two test models.

```mo
model Test_CubicInterpolation_lambda "Direct formulation"
  parameter Real Re1=2000;
  parameter Real Re2=4000;
  parameter Real delta=0;
  Real Re=2000*(1 + time);
  Real lambda2=Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_lambda(Re, Re1, Re2, delta);
end Test_CubicInterpolation_lambda;
```

gives the blue curve in

[image]

and

```mo
model Test_CubicInterpolation_Re "Inverse formulation"
  parameter Real Re1=2000;
  parameter Real Re2=4000;
  parameter Real delta=0;
  Real lambda2=1000*125*(1 + 4.2*time);
  Real Re=Modelica.Fluid.Dissipation.Utilities.Functions.General.CubicInterpolation_Re(0, Re1, Re2, delta, lambda2);
end Test_CubicInterpolation_Re;
```

gives the red curve. This matches the descriptions and diagrams in [Modelica.Fluid.UsersGuide.ComponentDefinition.WallFriction](https://doc.modelica.org/Modelica%203.2.3/Resources/helpDymola/Modelica_Fluid_UsersGuide_ComponentDefinition.html#Modelica.Fluid.UsersGuide.ComponentDefinition.WallFriction), though I wonder why the two signals do not intersect at Re=Re2. Also, it is odd that the first input argument `Re_turbulent` of CubicInterpolation_Re is not used within the function.

beutlich, 2019-10-29:
We could also utilize the Modelica.Fluid.Utilities.cubicHermite function in the interpolation functions for the modified wall friction coefficient lambda2 and the Reynolds number. Though it gives less optimized code, it is easier to comprehend. Here is the full example which gives the identical result as above.

```mo
model Test_CubicInterpolation_lambda_2 "Direct formulation 2"
  parameter Real Re1=2000;
  parameter Real Re2=4000;
  parameter Real delta=0;
  Real Re=2000*(1 + time);
  Real lambda2=CubicInterpolation_lambda_2(Re, Re1, Re2, delta);
  function CubicInterpolation_lambda_2
    extends Modelica.Icons.Function;
    import Modelica.Math;
    import Modelica.SIunits;
    input SIunits.ReynoldsNumber Re;
    input SIunits.ReynoldsNumber Re1;
    input SIunits.ReynoldsNumber Re2;
    input Real Delta;
    output Real lambda2;
    // point lg(lambda2(Re1)) with derivative at lg(Re1)
  protected
    Real x1=Math.log10(Re1);
    Real y1=Math.log10(64*Re1);
    Real yd1=1;

    // Point lg(lambda2(Re2)) with derivative at lg(Re2)
    Real aux1=(0.5/Math.log(10))*5.74*0.9;
    Real aux2=Delta/3.7 + 5.74/Re2^0.9;
    Real aux3=Math.log10(aux2);
    Real L2=0.25*(Re2/aux3)^2;
    Real aux4=2.51/sqrt(L2) + 0.27*Delta;
    Real aux5=-2*sqrt(L2)*Math.log10(aux4);
    Real x2=Math.log10(Re2);
    Real y2=Math.log10(L2);
    Real yd2=2 + 4*aux1/(aux2*aux3*(Re2)^0.9);

  algorithm
    lambda2 := 10^Modelica.Fluid.Utilities.cubicHermite(Math.log10(Re), x1, x2, y1, y2, yd1, yd2);
    annotation (Inline=false, smoothOrder=5);
  end CubicInterpolation_lambda_2;
end Test_CubicInterpolation_lambda_2;
```

I also recapped and confirmed the interpolation formula for lambda2 as questioned by @dofware, see 
[CubicInterpolationMoody.pdf](https://github.com/modelica/ModelicaStandardLibrary/files/3786076/CubicInterpolationMoody.pdf).

Nevertheless, I am going to

* improve description comments and documentation
* add the alternative implementation using Modelica.Fluid.Utilities.cubicHermite at least as comment
* change the integer literal `1` to `yd1` (to make the formula more explicit)

in a subsequent pull request.

PR #3185 (2019-11-18): Add documentation and description strings to Modelica.Fluid.Dissipation.Utilities.Functions.General.{CubicInterpolation_lambda, CubicInterpolation_Re}
* Improve description comments and documentation
* Add the alternative implementation using Modelica.Fluid.Utilities.cubicHermite at least as comment
* Change the integer literal `1` to `yd1` (to make the current optimized formula more explicit)

resolves #2972
closed 2019-11-18 by beutlich