modelica/ModelicaStandardLibrary #3002: Purpose of Modelica.Math.Vectors.Utilities.householder{Vector,Reflection)?

HansOlsson, 2019-06-24:
Based on looking at PR #2986 I noticed these two functions that are not used anywhere in MSL.

Both of them also contain odd statements, that indicate that some additional documentation is needed - in particular:
* householderReflection sets small element to zero. I don't know why; and I don't believe that to be usual. It could be that it is relevant for some application, but there is no indication of that application would be.
* householderVector has some odd code for alpha - I believe it is related to avoiding round-off errors (as discussed in Golub, van Loan, "Matrix Computations"), but there is neither a reference nor a comment.

beutlich, 2019-06-24:
They are used in ModelicaTest at least.

henrikt-ma, 2019-06-24:
I would be in favor of removing this sort of functions from MSL.  I don't really see how they fit into the world of Modelica model computations.  I do see how they fit in a surrounding scripting environment, but in my mind, the MSL shouldn't be a container for scripting environment functions.  I also see how Householder reflections play a role in numeric solving of equation systems, but then the use should be hidden inside the tools and their simulation runtime libraries, rather than direct invocation from Modelica user code.

beutlich, 2019-06-24:
Yes, we could move these two functions to ObsoleteModelica4 and completely get rid of Modelica.Math.Vectors.Utilities if we also move Modelica.Math.Vectors.Utilities.roots to Modelica.Math.Polynomials.

MartinOtter, 2019-07-19:
Just for information: The purpose of the numerical functions in Vectors and Matrices is to use them in a controller that is used in a model (so it is part of model and therefore post-processing functionality does not help here). For example, one could design a Riccati controller at every sample point. I do not recall why householderVector and householderReflection are in MSL. The intention was to move more of the numerical functions for controller design from Modelica_LinearSystems2 to MSL, but this was only partially done (e.g. Matrices/balanceABC, discreteRiccati etc.)

beutlich, 2019-07-19:
So you you agree to move householderVector and householderReflection to ObsoleteModelica4 (as they are still available from Modelica_LinearSystems2.Math.Vectors)? I'd also like to move Modelica.Math.Vectors.Utilities.roots to Modelica.Math.Polynomials then.

AHaumer, 2019-07-19:
I would leave it to @MartinOtter @DrMartinKuhn and (I believe) @tobolar - AFAIK they know Modelica_LinearSystems2 best - to move those functions to Modelica_LinearSystems2 to avoid broken models.

MartinOtter, 2019-07-22:
> 
> 
> So do you agree to move householderVector and householderReflection to ObsoleteModelica4 (as they are still available from Modelica_LinearSystems2.Math.Vectors)? I'd also like to move Modelica.Math.Vectors.Utilities.roots to Modelica.Math.Polynomials then.

Fine for me (I have no strong opinion).

beutlich, 2019-07-29:
When working on this issue, I noticed that the mentioned functions from Vectors.Utilities are referenced from the documentation of Modelica.Math.Matrices.Utilities.householderReflection and Modelica.Math.Matrices.Utilities.householderSimilarityTransformation. Should these two functions also be moved to ObsoleteModelica since they are nowhere used and still available from  Modelica_LinearSystems2?

beutlich, 2019-09-05:
> When working on this issue, I noticed that the mentioned functions from Vectors.Utilities are referenced from the documentation of Modelica.Math.Matrices.Utilities.householderReflection and [...]

@MartinOtter Any feedback?

PR #3120 (2019-09-16): Remove special householder transformation functions
Functions are still available via ObsoleteModelica4.Math or [Modelica_LinearSystems2.Math](https://github.com/modelica/Modelica_LinearSystems2).

Close #3002.
closed 2019-09-16 by beutlich