modelica/ModelicaStandardLibrary #3021: Power equations

AHaumer, 2019-07-02:
A question to @christiankral , @dietmarw , @beutlich ,  @HansOlsson :
Could we replace the power equations in Modelica.Electrical.QuasiStatic.SinglePhase.Interfaces.TwoPin as shown here 
[power.txt](https://github.com/modelica/ModelicaStandardLibrary/files/3351976/power.txt), with the following changes for users, but increasing clarity:
S -> abs_S
P -> S.re
Q -> S.Im
This is not backwards compatible - could that be handled by a conversion script?
These changes would be applied analogously to Modelica.Electrical.QuasiStatic.MulitPhase.Interfaces.TwoPlug and Modelica.Electrical.QuasiStatic.MulitPhase.Interfaces.OnePort.

AHaumer, 2019-07-04:
The proposal handles (complex) power the same way as voltages and currents. 
Could please anyone give a hint whether the above mentioned changes can be handled by a conversion script?

AHaumer, 2019-07-04:
In case of polyphase, we would use ComplexApparentPower S[m] and S_total, i.e.
S[m] -> abs_S[m]
P[m] -> abs_S[m].re
Q[m] -> abs.S[m].im
S_total -> abs_S_total
P_total -> S_total.re
Q_total -> S_total.im

beutlich, 2019-07-08:
@HansOlsson Is such kind of renming possible by convertClass?

HansOlsson, 2019-07-08:
> In case of polyphase, we would use ComplexApparentPower S[m] and S_total, i.e.
> S[m] -> abs_S[m]
> P[m] -> abs_S[m].re

To clarify: You mean that this applies to any index 'm', right?

beutlich, 2019-07-08:
Hm, I tought that array component P should get renamed to abs_S.re.

HansOlsson, 2019-07-08:
> Hm, I tought that array component P should get renamed to abs_S.re.

Except that as I understand it abs_S is the new array.

beutlich, 2019-07-08:
Yes, basically it is deleting and adding variables. Is that possible by conversion support?

Old:
```mo
model M "Old model"
  parameter Integer m=3;
  Complex X[m] = fill(Complex(1, 1), m);
  Real P[m] = X.re;
  annotation(uses(Complex="3.2.3"));
end M;
```

New:
```mo
model M "New model"
  parameter Integer m=3;
  Complex X[m] = fill(Complex(1, 1), m);
  Real abs_S[m] = {(c.re^2 + c.im^2)^0.5 for c in X}; // No longer connected to P from old model.
  annotation(uses(Complex="4.0.0-dev"));
end M;
```

> In case of polyphase, we would use ComplexApparentPower S[m] and S_total, i.e.
> P[m] -> abs_S[m].re
> Q[m] -> abs.S[m].im

abs_S is Real, right?

HansOlsson, 2019-07-08:
I thought that in the example above the idea was to convert `P[m]` to `X[m].re` and in general any element of this array in this way, since P was removed. However, I might be mistaken, so it would be good to have it clarified.

beutlich, 2019-07-08:
As I understood it, at least for the single-phase case:
   - Remove Reals P, Q and S.
   - Add Reals abs_S, arg_S and Complex S.
This means that the type of S changes from Real to Complex.

HansOlsson, 2019-07-11:
> As I understood it, at least for the single-phase case:
> 
> Remove Reals P, Q and S.
> Add Reals abs_S, arg_S and Complex S.
> 
> This means that the type of S changes from Real to Complex.

It should be handled, and **ideally** it shouldn't matter that both are called 'S'.

However, in case of incorrect implementation (and/or confused users) the conversion may run more than once and that will cause a completely mess - so there is a minor advantages in having different names. Thus if we can find a more descriptive new name than 'S' I would think that is good.

dietmarw, 2019-07-25:
> A question to @christiankral , @dietmarw , @beutlich , @HansOlsson :
> Could we replace the power equations in Modelica.Electrical.QuasiStatic.SinglePhase.Interfaces.TwoPin as shown here [...]

@AHaumer  I'd say the new way would confuse the user more than using S, P and Q? Could you explain why you think that using the complex parts would make it easier to understand the equations? I'd have thought the opposite.

AHaumer, 2019-07-26:
@dietmarw 
If you consider extending the implementation to harmonics (LTI-system), you get in troubles with variable names. It would be much clearer to have Complex apparent power per harmonic (i.e. an array), knowing that the imaginary part is reactive power due to phase shift of the reactive harmonic. 
Total apparent power is the complex sum of the array of apparent powers related to the harmonics. Active power is the real part, but the imaginary part is only reactive power due to the phase shifts, reactive power due to distortion (from purely sinusoidal waveform) has to be added (respectively is included in total current rms times total voltage rms).
It's also a question of consistency: For voltages and current swe look after the complex values, knowing the real parts are active and the imaginary parts are reactive voltages or currents.

christiankral, 2019-10-07:
I have no strong opinion on this subject. I think that both ways on spelling active, reactive apparent power have pros and cons. However, I think that new Modelica users find it easier to understand `P`, `Q` and `S` instead of `S.re`, `S.im` and `abs_S`.

dietmarw, 2019-10-07:
So I'd suggest to leave them as S, P and Q, especially for beginner's sake. Is that OK for you @AHaumer? In which case we could close this?

christiankral, 2020-03-01:
I am closing this ticket as I assume agreement from @AHaumer .

closed 2020-03-01 by christiankral