modelica/ModelicaStandardLibrary #3022: Unify names of electrical signal outputs

christiankral, 2019-07-03:
In the different domains we have different names of the signal output connectors. For example, the voltage sensor:

Sensor | Output connector name | Comment
---------- | ------------------------------ | --------------
`Analog.Sensors.VoltageSensor` | `v` | Instantaneous voltage, lower case  
`Machines.Sensors.VoltageQuasiRMSSensor` | `V` | Quasi RMS voltage, upper case
`QuasiStatic.SinglePhase.Sensors.VoltageSensor` | `y` | Complex quasi static
``
`QuasiStatic.MultiPhase.Sensors.VoltageSensor` | `y[m]` | Complex qusai static vector
`QuasiStatic.MultiPhase.Sensors.VoltageQuasiRMSSensor` | `V` | Complex quasi RMS voltage, upper case 

So this is a very non-uniform look. Change proposals
1. Change `y` to `v` and `y[m]` to `v[m]`
   * Pro: a voltage is clearly indicated as voltage `v`, characteristic voltage quantities are indicated by upper case `V`
   * Con: Partial sensors with `RealOutput y` cannot be re-used
2. Change every output to `y`
    * Pro: Clear rule: `y` is always the output of any sensor
    * Con: From the connector output name it cannot be told what kind of quantity it is

AHaumer, 2019-07-04:
We can discuss whether to use an uppercase V for RMS voltage output. 
But @christiankral is right, "y" should be "v". 
And I think we should add units!

AHaumer, 2019-07-04:
@christiankral  I vote for proposal 1.

christiankral, 2019-07-17:
I started to work on the changes introduced by proposal 1 and ran into some issue I have no solution currently for. Consider the partial sensor model of the quasi-static single-phase library, which at the end relies on the partial model `Modelica.Electrical.QuasiStatic.SinglePhase.Interfaces.TwoPin`. In this model partial model a lot of characteristic quantities are defined, including the voltage `v` and the current `i`. This model serves as code basis for the components, sources and sensors. 

```modelica
partial model TwoPin "Two pins"
  import Modelica.Constants.eps;
  Modelica.SIunits.ComplexVoltage v "Complex voltage";
  Modelica.SIunits.Voltage abs_v=Modelica.ComplexMath.'abs'(v) "Magnitude of complex voltage";
  Modelica.SIunits.Angle arg_v=Modelica.ComplexMath.arg(v) "Argument of complex voltage";
  Modelica.SIunits.ComplexCurrent i "Complex current";
  Modelica.SIunits.Current abs_i=Modelica.ComplexMath.'abs'(i) "Magnitude of complex current";
  Modelica.SIunits.Angle arg_i=Modelica.ComplexMath.arg(i) "Argument of complex current";
  Modelica.SIunits.ActivePower P=Modelica.ComplexMath.real(v*
      Modelica.ComplexMath.conj(i)) "Active power";
  Modelica.SIunits.ReactivePower Q=Modelica.ComplexMath.imag(v*
      Modelica.ComplexMath.conj(i)) "Reactive power";
  Modelica.SIunits.ApparentPower S=Modelica.ComplexMath.'abs'(v*
      Modelica.ComplexMath.conj(i)) "Magnitude of complex apparent power";
  Real pf=cos(Modelica.ComplexMath.arg(Complex(P, Q))) "Power factor";
  Modelica.SIunits.AngularVelocity omega "Angular velocity of reference frame";

  PositivePin pin_p "Positive quasi-static single-phase pin" annotation (...);
  NegativePin pin_n "Negative quasi-static single-phase pin" annotation (...);
equation
  Connections.branch(pin_p.reference, pin_n.reference);
  pin_p.reference.gamma = pin_n.reference.gamma;
  omega = der(pin_p.reference.gamma);
  v = pin_p.v - pin_n.v;
  i = pin_p.i;
end TwoPin;
```

In the current implementation the partial sensor model adds the complex output `y`:

```modelica
partial model RelativeSensor "Partial sensor"
  extends Modelica.Icons.RotationalSensor;
  extends OnePort; // which entends TwoPin and adds pin_p.i + pin_n.i = Complex(0);
  Modelica.ComplexBlocks.Interfaces.ComplexOutput y annotation (...);
end RelativeSensor;
```

In proposal 1 the plan is to make 

* `v` the `ComplexOutput` of the voltage sensor
* `i` the `ComplexOutput` of the current sensor

How can this be done without re-writing the code for each of the two sensors and without being too difficult so that a Modelica beginner can still understand how the different sensors work? Comments and suggestions are very welcome.

christiankral, 2019-07-17:
**Note 1.** To my knowledge, we currently have in the MSL two ways of how sensor outputs are named. These two ways are summarized by proposal 1 and 2 in the first entry of this ticket (https://github.com/modelica/ModelicaStandardLibrary/issues/3022#issue-463752572).

* **Proposal 1** will always be more difficult to be implemented as partial models cannot simply be re-used. However, I am still in favor of proposal 1.
* **Proposal 2** has very high degree of modularity. This approach could even be extended to `Sources` where all the inputs of single input sources could be named `u`. This naming scheme would then start to cause inconsistencies when more and one signal inputs (in sources) and outputs (in sensors) are used, because `u1`, `u2` and `y1`, `y2` were definitely bad naming options.

**Note 2.** The signal voltage and current source currently avoid this problem by naming the signal inputs with uppercase `V` and `I`. This thought, however, were a third option to name the sensor outputs. As the intention of this ticket was to unify the naming of the electrical sensors -- and possibly, in a second step also include other sensors -- it makes sense to thoroughly evaluate different design options.   

**Note 3.** As a conclusion of note 2 we possibly also have to think about unifying the naming so source inputs.

AHaumer, 2019-07-17:
Just a remark to uppercase and lowercase, e.g. V and v:
Normally in Modelica we use uppercase for parameters and lowercase for time varying variables. 
In Engineering we use uppercase for constant values or processed values e.g. RMS and lowercase for time varying variables.

As a consequence, if a sensor output is an unprocessed time signal, it should be lowercase. 
If a sensor output is a processed signal e.g. RMS, it should be uppercase.

A special case is Electrical.QuasiStatic:
In Engineering, we use an underlined uppercase for a complex time phasor. 
In Modelica, we used lowercase since the complex phasors may vary with time. 
So, using an uppercase as sensor output would be a consistent solution (and easy to understand, only the underline for complex is missing).

A further remark on Electrical.Analog.Sensors:
The sensors nclude the same balance equations as OnePort!
Should we change that, i.e. sensors also extending from partial OnePort?
Then we would have to change the output, too, since we cant change the inherited variable e.g. v to an output.

Conclusion:
We should find a nice consistent naming of signal inputs and signal outputs for pyhical variables, bearing a hint which varianle is mant in the name, and, of course, having the attribute "unit".
"u" for input and "y" for output is fine for Blocks only.
Just an idea (maybe this triggers even better proposals):
prefix or suffix "in for inputs and prefix of suffix "out" for outputs, e,g,
v -> in_v / v_in
v -> out_v / v_out

christiankral, 2019-09-21:
Having an awkward name such as `out_v` or `v_out` in a sensor does not convince me. Unfortunately we do not have any alternative proposals to be considered. Therefore, for the easiness of implementation the simplest way may be to go for proposal 2:

- Simple Modelica implementation
- Easy to understand even for Modelica beginners
- Easy nomenclature for the naming of signal outputs: the name is always `y` for sensor with one signal output

tobolar, 2019-10-01:
> simplest way may be to go for proposal 2

This will definitely concern also other domains. In this case, more library officers shall be involved.

christiankral, 2019-10-07:
I agree that we should discuss this on a broader level as it makes sense to implement the unification for all MSL packages.

MartinOtter, 2019-10-16:
I did not try out, but it should be possible to redeclare the declaration of "v" or "i" in TwoPin by adding a prefix output, something like (maybe @HansOlsson can help with the exact syntax):
```
 extends TwoPin(redeclare output Modelica.SIunits.ComplexVoltage v annotation(...));
```

christiankral, 2019-10-29:
I tried a couple variants but could not find a proper solution. @HansOlsson do you have an idea how this could be implemented?

HansOlsson, 2019-10-30:
> I did not try out, but it should be possible to redeclare the declaration of "v" or "i" in TwoPin by adding a prefix output, something like (maybe @HansOlsson can help with the exact syntax):
>  extends TwoPin(redeclare output Modelica.SIunits.ComplexVoltage v annotation(...));

I can't find one either, I can only see one possible legal one - which is messier than the ideas above, but will need to investigate more (and it is anyway not fully implemented in Dymola).

It would still not be intuitive, and I would instead propose that we instead improve the situation in another way: TwoPin contains too much in my opinion, and it would be helpful to have a minimal TwoPin that only has the two pins (and in this domain the connection between the reference-angles); and similarly for OnePort.

If we had such a minimal OnePort it would be more straightforward since we wouldn't have to change any internal ComplexVoltage to output connector.

christiankral, 2019-10-30:
So it seems we can only have either one or the other option:

**Option 1:** Modular concept with output `y` (proposal 1)
**Option 2:** Far less modular concept (proposal 2) where we re-write the following equations for each sensor:

```modelica
  Modelica.SIunits.Voltage abs_v=Modelica.ComplexMath.'abs'(v) "Magnitude of complex voltage";
  Modelica.SIunits.Angle arg_v=Modelica.ComplexMath.arg(v) "Argument of complex voltage";
  // Modelica.SIunits.ComplexCurrent i "Complex current";
  Modelica.SIunits.Current abs_i=Modelica.ComplexMath.'abs'(i) "Magnitude of complex current";
  Modelica.SIunits.Angle arg_i=Modelica.ComplexMath.arg(i) "Argument of complex current";
  Modelica.SIunits.ActivePower P=Modelica.ComplexMath.real(v*
      Modelica.ComplexMath.conj(i)) "Active power";
  Modelica.SIunits.ReactivePower Q=Modelica.ComplexMath.imag(v*
      Modelica.ComplexMath.conj(i)) "Reactive power";
  Modelica.SIunits.ApparentPower S=Modelica.ComplexMath.'abs'(v*
      Modelica.ComplexMath.conj(i)) "Magnitude of complex apparent power";
  Real pf=cos(Modelica.ComplexMath.arg(Complex(P, Q))) "Power factor";
```

I have, however, to admit that, for example, the current sensor does not have to calculate the magnitude and the angle of the complex voltage or the complex apparent power, etc., as those things are not relevant at all. So if we go for proposal 2 

> * **Proposal 1** will always be more difficult to be implemented as partial models cannot simply be re-used. However, I am still in favor of proposal 1.
> * **Proposal 2** has very high degree of [...]

option 2 may be right choice.

HansOlsson, 2019-10-31:
To me the voltage-sensor becomes easier to understand if we remove the unnecessary variables (like magnitude, angle, apparent power etc).

One of the issues is that we stuff so much functionality into the components that it may create a barrier for new users.

christiankral, 2020-01-19:
I am working on a PR to fix this issue

beutlich, 2020-01-20:
> 
> 
> I am working on a PR to fix this issue

Is this a MSL-wide issue or does it only apply to Electrical.QuasiStatic?

christiankral, 2020-01-20:
This issue only applies to `Electrical.QuasiStatic` -- to my knowledge. Mechanical, Magnetic, Thermal,... they do not use `y`, but the respective physical quantity instead.

PR #3351 (2020-01-21): Rename sensor outputs
This PR resolves #3022. The following changes are applied both in `QuasiStatic.SinglePhase` and `QuasiStatic.Polyphase`:

- Rename all sensor outputs to match their physical quantity (e.g., `i` for current sensor)
- Introduce new partial models `TwoPinElementary` with connectors but without declaration of voltage and current
- Introduce new partial model `RelativeSensorElementary` to be used with voltage and current sensor(s only)
- Move obsolete partial models `RelativeSensor` to `ObsoleteModelica4.mo` as these models have only been used for voltage and current sensors;  [... cut]
closed 2020-01-21 by beutlich