modelica/ModelicaStandardLibrary #3034: Need to have a reactive power sensor

max-privato, 2019-07-12:
I think the availability of a good reactive power sensor would be very useful for analysing single-phase quasi-sinusoidal circuits.
I've a simple component ("QSensor") that I created several years ago and I've been using since then satisfactorily. In the enclosed zip I've put this component and a simple circuit for testing it. The two models contain some description (Documentation annotation) already.

[Files.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3385281/Files.zip)

christiankral, 2019-07-12:
So the proposed model is based on the a `FixedDelay` block:

[image]

AHaumer, 2019-07-12:
Sorry, I'm **strictly against** using a delay. Delay might cause problems in some tools. Are events also delayed? It's a similar problem as with the mean-block (which also has a delay). 
Both solutions depend on a **fixed** frequency. If you have a system with variable frequency, you're lost.
I know that the solution for polyphase systems is easier, and it seems that you want to have a single-phase solution. We can discuss that and search for a robust solution, but I warn of using such things like delay or mean-block inside a sensor. Always use such blocks outside a sensor! 
We could cosider the following:
Feed the voltage and current sensor signal to a "signal processing power analyzer" block. 
In this block we could try to calculate harmonics (maybe by fft? or by harmonic-blocks). 
In fact the whole problems (at least for single-phase) needs more inverstigation and discussion.

christiankral, 2019-07-12:
All the electrical sensor of the Modelica Standard Library contain no signal processing and therefore the sensor output signals have no time delay at all. Some sensor may contain calculations, but they all rely on on instantaneous calculations. I am therefore also not in favor of including such reactive power sensor. 

Instead I provide a small Modelica library [ElectricalSignalProcessing](https://gitlab.com/christiankral/ElectricalSignalProcessing). This library now contains only one block to process the active, reactive and apparent power (and the power factor) with respect to one harmonic ordinal number `k`. This block works strictly without `FixedDelay`, but processes `Modelica.Blocks.Math.Harmonic` instead. A test example is also added.

max-privato, 2019-07-12:
_Both solutions depend on a fixed frequency. If you have a system with variable frequency, you're lost._
Yes. Indeed I evidenced the fixed frequency in the icon exactly because of this.
Before proposing this sensor I've been using it for years for teaching, and I really found it useful. It works only within fixed frequency electric circuits.

_I know that the solution for polyphase systems is easier, and it seems that you want to have a single-phase solution._ 
Yes, because we have constant output for balanced systems. But you are right I proposed a single-phase option.

max-privato, 2019-07-12:
_All the electrical sensor of the Modelica Standard Library contain no signal processing and therefore the sensor output signals have no time delay at all. Some sensor may contain calculations, but they all rely on on instantaneous calculations. I am therefore also not in favor of including such reactive power sensor._
If this is the philosophy, obviously my proposal is against it. I made my proposal since I was not aware of this philosophy. After all I don't see any big problem in using Mean, nor a sensor with some dynamics (as real-life ones).

_Instead I provide a small Modelica library ElectricalSignalProcessing. This library now contains only one block to process the active, reactive and apparent power (and the power factor) with respect to one harmonic ordinal number k. This block works strictly without FixedDelay, but processes Modelica.Blocks.Math.Harmonic instead. A test example is also added._
This is nice, but in my opinion could be overkill for several simpler circuits. This model also, as mine, requires fixed frequency circuits. However, I think some Q sensor is well needed. If this one, or some evolution of it, is finally accepted I'll ve glad with this decision.

AHaumer, 2019-07-13:
If we deal only with fixed frequency:
Why not choose Electrical.QuasiStationary (renamed in MSL 4 to QuasiStatic)?
Reactive power Q is well defined ...
In case of a transient simulation (like starting an induction machine), I wonder what the Q-sensor delivers.

christiankral, 2020-02-15:
As we already have the `Modelica.Electrical.Polyphase.Sensors.ReactivePowerSensor` for three-phase systems and the library [ElectricalSignalProcessing](https://gitlab.com/christiankral/ElectricalSignalProcessing) was provided, I am closing this ticket.

closed 2020-02-15 by christiankral