modelica/ModelicaStandardLibrary #3050: Error reading mat file in Modelica.Blocks.Tables.CombiTable1D

jbgohl, 2019-07-15:
The attached mat file "my_test.mat" was created using the attached Python script "test_scipy.py" using Anaconda python and scipy version 1.2.1.  When I try to simulate the attached model "test3.mo" I get the following error during initialization:

    InflateFieldNames: inflate returned stream end

I've tested that I get the same error in Dymola 2018 FD01 (Modelica version 3.2.2, 2016-04-03, build 3 (2016-04-03 08:44:41Z)), 2020 (Modelica versions 3.2.3, 2019-01-23, build 2 (2019-03-20 12:00:00Z) and 4.0.0-dev, 2019-01-23, build 1 (2019-01-23 12:00:00Z), with appropriate class name changes) and JModelica.

I can load my_test.mat without error into both MATLAB (version 2015b) and GNU Octave (version 5.1.0).  If I re-save the mat file (using the following command) from either MATLAB or Octave, the model test3.mo then simulates without error.

    load my_test.mat
    save -v6 my_test.mat struct1

I've also tried disabling compression when saving the file with "do_compression=False".  In that case the simulation still fails but the error message is different:

    fields[0] not MAT_T_MATRIX, fpos = 784

Can anyone comment if this is an error with the mat file, as created by scipy, or a limitation of the ModelicaMatIO code, (or user error)?

[190626.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3393950/190626.zip)

beutlich, 2019-07-15:
Can reproduce with matdump 1.516. Looks like an upstream issue for matio.

Can you please attach the MAT-file created in MATLAB by

```
load my_test.mat
save -v7 my_test_v7.mat struct1
```

jbgohl, 2019-07-15:
Here are the versions created with both v6 and v7.
[my_test_v67.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3394009/my_test_v67.zip)

beutlich, 2019-07-15:
Thanks. Files my_test_v7.mat differs from my_test.mat (from scipy) starting at byte 290. If I load and save the my_test_v7.mat directly in matio, it also is different again from byte 290. The issue is in reading the field name length of struct1.test4 (being a struct itself and last in the file). Need to analyze the stream inflation in detail and compare why it works for the files saved in MATLAB or matio, but fails for the scipy one.

beutlich, 2019-07-15:
I'd also need the MAT-file save in scipy with `do_compression=False`, since the stream issue just is a subsequent error (of the orignal data). Thanks again.

beutlich, 2019-07-15:
Just installed scipy 1.3.0 and create it on my own.

jbgohl, 2019-07-15:
Here are the versions created with scipy 1.2.1, both uncompressed (..._u.mat) and compressed (..._c.mat).
[my_test_uc.zip](https://github.com/modelica/ModelicaStandardLibrary/files/3394308/my_test_uc.zip)

beutlich, 2019-07-15:
OK, I now know what the problem is. The MAT-file specification optionally enables storage optimization for variable names if they fit within 4 bytes. In your file, this is the case for the field names a and b of struct1.test4 and scipy uses this kind of optimization when saving to v5 MAT-file (no mater if compressed or not). This optimization is not (yet) supported for field names of structs by function ReadNextStructField of matio.

Up to now, this issue never arose since MATLAB always stores the struct field names in the long variant (even four bytes in for each of the names a and b). matio does the same when storing field names (but uses three bytes for each of the names a and b). This also is the reason why the MAT-files differ in size.

Let me see when I can fix it and provide new maintenance releases of both matio and ModelicaMatIO.

As workaround you can go ahead if you force longer field names for scipy.io.savemat, i.e., changing line 15 of your script from

```py
a['struct1']['test4']['a']=20.123
```

to

```py
a['struct1']['test4']['a1']=20.123 # increase field name size from 1 to 2 bytes (for all fields)
```
 circumvents the issue for now.

jbgohl, 2019-07-15:
Confirmed.  Increasing the name from "a" and "b" to "a1" and "b1" worked.  Thanks!

PR #3006 (2019-07-23): Update Matio library v1.5.17
Superseding #2877. Closes #2905, closes #3050.
closed 2019-07-23 by beutlich